winder/Universal-G-Code-Sender #1052: COM port not found - UGS Platform on Mac

ntewinkel, 2018-06-12:
## Problem description 
When running UGS Platform on Mac, the software claims "no serial ports found", but Arduino IDE shows the serial port for the CNC machine is there.

### Expected Behavior
Serial port should show in dropdown list, and be selectable

### Actual Behavior
Port list is empty

### Steps to Reproduce the Problem
Connect CNC machine with USB cable
start up UGS Platform
At top, by "Port", press the refresh button (app says "no serial ports found.")
Also the dropdown picklist box remains empty.

### Specifications
#### Version
Nightly build, November 11. Version 2.0

#### Operating system
 Mac OSX  High Sierra

#### Platform
LinkSprite CNC machine (pre-assembled), with Woodpecker CNC GRBL board.
Com port shows up in Arduino as /dev/tty.wchusbserial620

Similar to issue reported last year for original UGS:
https://github.com/winder/Universal-G-Code-Sender/issues/599

Note that old workaround of pasting the com port into the text box is not possible in UGS Platform.
I was able to use the workaround to get UGS classic working with my CNC machine.
Note that issue #599 is closed, but the issue still persists in the classic version as well.

breiler, 2018-06-12:
This is an known error which we might have a workaround for.

Update to the latest nightly build then follow the instructions at the bottom here: #940

Let us know if this works for you.

ntewinkel, 2018-06-12:
Ah yes! that works, thank you breiler.
For others viewing this, this is what I did:
For UGS Platform (Nov 11 build)
Preferences, choose UGS button/tab, then choose Connection Driver = JSerialComm

ntewinkel, 2018-06-12:
ps, maybe the "no serial ports found" message could add a reference to that connection driver option?

breiler, 2018-06-13:
I'm hoping that someone using Windows could test this driver. If that works we can replace the old driver all together.

Thanks for the feedback!

closed 2018-06-13 by breiler

nvaken, 2019-06-13:
I'm afraid that the option "Connection Driver" is no longer to be found under Sender options?

JacksCrafts, 2019-07-20:
Hi Kevin,
You might try using a newer version?
There have been a lot of improvements.

Product Version: Universal Gcode Platform 2019 04 23
Java: 1.8.0_211; Java HotSpot(TM) 64-Bit Server VM 25.211-b12
Runtime: Java(TM) SE Runtime Environment 1.8.0_211-b12
System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)

Kevinvs7, 2019-07-20:
Hey man!

Thank you for your reply. I downloaded UGS like two days ago but I will try to remove and download all again and see if it works. Thank you once again!

Kevin