winder/Universal-G-Code-Sender #1065: pausing feature request

ukoksoy, 2018-06-19:
Hi. I am doing wood carving using arduino grbl. Some operations take more than 3 hours. I sometimes need to pause the operation, shut down the computer, power off the arduino and continue the job from where I left in another day. Current pause pause feature works well if I dont disconnect arduino. It would be great if the software saves the last point when pause button is clicked. After restarting computer and ugs, a confirmation box may appear asking to continue the paused operation.

zeevy, 2018-06-20:
I don't think this is possible in grbl world any time soon, you need to divide your job in multiple parts

ukoksoy, 2018-06-20:
I think it is possible. When you click the pause button, ugs can save the remaining gcode as a different gcode file in the computer. And when you click the resume button, it can run that file.
This is what I think. May be I am wrong.

zeevy, 2018-06-20:
Can't save the remaining gcode file alone, as it needs grbl status like (feed, units, other status). And its bit tricky to save all those information and resume from there. 
There is skip n lines feature (which is pretty close to your requirement) seems to be implemented on laser grbl gui, where you can skip n number of lines in the gcode file
https://github.com/gnea/grbl/issues/305

ukoksoy, 2018-06-23:
Ok then. When I pause the operation, I copy the last gcode in UGS console. After that I manually update update my gcode file by removing previous gcodes but include feed, unit etc settings. It works fine. 
But I use UGS arc expander setting which converts G2/G3 commands to G1 command.
When I pause the operation, last gcode command is G1 command which is already converted by UGS and I cant see that command in my original gcode file.
Is there anything that I can do to see the last command as G2/G3 command?

breiler, 2020-11-01:
Closing as it now is possible to start from a given line number in UGS.

closed 2020-11-01 by breiler

SharpiePanick, 2021-03-19:
I can’t this feature or any other mention of it anywhere?! Where can it be found?

breiler, 2021-03-19:
You are right, there is documentation missing on this. 
In the latest nightly build you can right click in the editor and choose "Run from":
[image]

SharpiePanick, 2021-03-20:
Alongside the pendant, best new feature ever. Thank you!