winder/Universal-G-Code-Sender #1070: Jog During Pause Feature

johnelle, 2018-06-29:
## Feature request

I just ran into a situation where some bolts came loose during the middle of a 6 hour cut and needed to be tightened.  Access was behind the Z carriage so to tighten them I had to jog to the Y edge, jog Z down to the bottom and then tighten them.  Unfortunately I had to completely restart the job because the jog buttons aren't available during a pause.   I assume that you have a known state for the flow so you could interject jog commands as long as everything returned to the position at pause.

zeevy, 2018-07-06:
I think grbl itself won't allow jog command during run or hold states, there is nothing gui can do on this unless it stops the current job and continue later from the last stopped position.

jahnj0584, 2018-09-18:
no CNC controller out there allows for this. Its just asking for safety concerns and troubles. Something like this just comes down to owner responsibility and doesnt have to be a software feature for such a rare and minute instance.

breiler, 2019-03-14:
See @chamnit:s answer in this forum:
http://www.shapeoko.com/forum/viewtopic.php?f=3&t=3822#wrap

> You can't move while you are paused mainly because of safety. Mills are much more dangerous than 3d printers, especially as you get bigger and cutting loads are not trivial. If you mess up, say jog up and around and return about 0.01" off of where you stopped, you're putting tremendous unintended loads on the cutter, which means it'll probably break. If you're hogging out material with a 1/2" cutter and it breaks, you do not want to be anywhere near it. 

I suggest that we close this issue?

closed 2019-04-20 by breiler