winder/Universal-G-Code-Sender #1071: Improve support for TinyG / g2core

breiler, 2018-07-02:
## Feature request
Improve the support for TinyG and g2core by adding or improve the support for:
* [ ] Streaming file, pausing and stopping
* [ ] Status report for work- and machine coordinates
* [ ] Jogging
* [ ] Homing
* [ ] Firmware settings

justinclift, 2018-07-06:
Ahhh, just came here to report "I can't seem to enable homing".  Turns out it's a known thing. :wink:

For me in testing now, using firmware version 101.03 (our latest edge build):

* Jogging is working
* Pause/resume is working
* Stopping works too
* Streaming a file seems to be working (did it with the motors off, and it's still running atm).

Homing works if done via the "Command" line (eg `G28.2 Z0 X0 Y0`), and the XYZ co-ordinates are correctly set when the home operation completes.

Noticed two bits of weirdness.  One is probably just me :wink:, while the other one might not be an issue or may need thinking about:

* "XY Step size" seems to be applied to Z movements too.  That wasn't obvious until trying it out and hitting my Z limit switch.  Should the title of that be "XYZ Step size" or is it expected to change the Z step size somewhere?
* Default feed rate (for me) was "10".  That's **super** slow.  With my Shapeoko 3 my normal feed rate is around 10-15k.  I don't remember if the feedrate is much lower with the stock Shapeoko 3 board though, as I haven't used that in ages.  It might well be.

Anyway, so far things look good.  Nothing terrible is jumping out at me anyway. :smile:

justinclift, 2018-07-06:
Hmmm, noticing one thing that I'm not sure about.  In the "Controller State (DRO)" Window, the line at the bottom looks like it's supposed to be the currently executing G-Code.  That's not changing for me atm, even though a file is being streamed.  The feed rate value and XYZ coordinates are changing dynamically though.

Should I take a screenshot and add it here?

justinclift, 2018-07-06:
#1075 updates the Help info to be more controller agnostic too.  Probably useful. :smile:

justinclift, 2018-07-06:
The file I was streaming (with motors off) finished successfully from the UGS perspective just now, so looks like that's good too. :smile:

justinclift, 2018-07-06:
> In the "Controller State (DRO)" Window, the line at the bottom looks like it's supposed to be the currently executing G-Code.

Ahhh, ignore that.  Just realised that's the "gcode state" mentioned in the welcome page tab.  No worries. :smile:

breiler, 2018-07-06:
> "XY Step size" seems to be applied to Z movements too. That wasn't obvious until trying it out and hitting my Z limit switch. Should the title of that be "XYZ Step size" or is it expected to change the Z step size somewhere?

This is my "fault". I created a new plugin for jogging that would scale better in the UI and provide continuous jogging (you can find it under Window -> Plugins -> Jog controller). I didn't want to mess with the existing working module. It's been around for a while so I mTo save space in the new plugin I decided we could use the same step size for XY and Z as default with the possibility to activate separate:
[image]

> Default feed rate (for me) was "10". That's super slow. With my Shapeoko 3 my normal feed rate is around 10-15k. I don't remember if the feedrate is much lower with the stock Shapeoko 3 board though, as I haven't used that in ages. It might well be.

That is correct, it's the default. 😃

justinclift, 2018-07-06:
Nifty, thanks. :smile:

justinclift, 2018-07-06:
Out of curiosity, is support for ABC (etc) axes in the forseeable future?

Just kind of thinking that might be useful, as it's one of the strengths of g2core compared to GRBL.

winder, 2018-07-06:
That's a work in progress hopefully I can keep finding time to work on it: https://github.com/winder/Universal-G-Code-Sender/issues/1068#issuecomment-402120810

justinclift, 2018-07-06:
Ooooo Aaaaaa!  Very nifty @winder. :smile:

I hadn't come across the grbl-Mega-5X project before either.  Looks like several microcontroller based systems are starting to bring "5 axes" to the general public then.  Hopefully at least one breaks through to getting it "mainstream". :smile:

justinclift, 2018-07-09:
[continuing a conversation from related #1072]

@scotthz With the testing I'm doing, that's been with 101.03 as well.  I'm not a java person either, so my knowledge of exactly what's needed is probably a bit iffy too.

The "UGS Platform" nightly build is the one I've tested.  After unpacking the .zip, I changed into the  `ugsplatform/bin` directory then launched `./ugsplatform`.  That "just worked", so I didn't need to muck around with any switches or similar to get it started.  This is on my CentOS 7 x64 desktop, using the OS provided OpenJDK v8:

* java-1.8.0-openjdk-1.8.0.171-8.b10.el7_5.x86_64
* java-1.8.0-openjdk-devel-1.8.0.171-8.b10.el7_5.x86_64
* java-1.8.0-openjdk-headless-1.8.0.171-8.b10.el7_5.x86_64

Those java packages were already installed on my desktop, but I'm not sure if that's because they're standard things or because some other previous thing I've used needed them.

Not sure how it'll all go on a Pi, but hoping it's ok. :smile:

scotthz, 2018-07-10:
In a nutshell, my experience was pretty much same as your own, @justinclift . I ran it on a Pi 3, because generally, that's what's attached to my machine, but I expect it would be the same running natively on my mac laptop.

I say "natively" because I tried to run it as a remote X11 client, with UGS running on the Pi and the UI running on my laptop, but that wasn't a lot of fun. It sort of works, but UGS doesn't appear to target this use case, and macOS isn't a great host for X11 clients in general. It makes the UI very slow and not suitable for a system that targets a realtime motion system. 

Being able to separate the UI from the "engine" is a key feature for me. This draws me to options like CNC.js, FabMo, and Chilipeppr. I'm ambivalent about Chilipepper. Some good ideas, but so javascript-centric that it's not all that fun if you're not already in that world. CNC.js has some of the same downside, but not as bad and I find it a lot more usable. FabMo seems to be a great idea, but too focused on ShopBot and/or too little interest in other platforms to seed much of a non-ShopBot community.

I am sure that UGS is pretty awesome for the GRBL world, but I am committed to G2, so for now, UGS is not compelling enough for **me** to want to use it given that it requires an attached computer, Pi or laptop, for the UI. Maybe when more of the basics are functional, like homing, WCS configuration, etc. I'd like to play more with the visualizer and other UI elements at that time.

breiler, 2018-07-10:
Thanks for giving this a try @scotthz, and you are right that running UGS like this is maybe not the primary use case. I'll keep working on the other base functions.

winder, 2018-07-10:
Thanks for the feedback.

The remote use case definitely isn’t a supported one. That said there is a web pendant that can be launched for remote access without sending the entire application. The “classic” UI is much lighter than the platform and may work better with that sort of configuration.

@scotthz Regarding the missing features, UGS has a controller abstraction layer so it is mainly a matter of implementing the missing features according to TinyG requirements. No feature in UGS is GRBL-Specific. Assuming the core TinyG operations work and people are interested, the remaining features are largely a [matter of filling in the blanks](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/TinyGController.java#L257). This includes the advanced widgest like probing and the configuration wizard.

justinclift, 2018-07-10:
For the filling in the blanks... does that mean things like the homing cycle are just a matter of making a PR with (say) something like?

```
sendCommandImmediately(new GcodeCommand("G28.2 Z0 X0 Y0"));
```

winder, 2018-07-10:
Exactly

justinclift, 2018-07-10:
Cool #1076 created.

Completely untested.  The G-Code itself is correct for XYZ homing though. :wink:

breiler, 2018-07-10:
I didn't want to add stuff that I couldn't properly try out for my self but I made the exact same code change (+ one extra line) that you made but got this error while trying it out and i didn't have time to look into it:
```
{"r":{"msg":"Z axis Homing Err - Homing input is misconfigured"},"f":[1,240,1]}
```

justinclift, 2018-07-10:
Ahhh, that just means your g2core settings aren't yet set up correctly.  It saying that it'd really like to do the homing thing, but isn't sure which digital inputs are used for each of your limit switches.

When you have a few minutes spare to look again, you're welcome to ping me and I can walk you through it if that helps. :smile:

justinclift, 2018-07-10:
As a thought, this **might** be a useful reference thing to look at:

* https://github.com/justinclift/g2/commit/ca4e95de8d589dfd22ca7b376d2d5685df73559e

They're the settings I use for my Shapeoko 3 here, which has X, Y, and Z limit switches.

Note the `#define Z_HOMING_INPUT              6` change on line 159.  Complete guess, but that's **probably** the kind of thing you'll need to change as well.

justinclift, 2018-07-10:
Excellent. :smile:

justinclift, 2018-07-15:
Just gave the most recent nightly builds (both Classic and Platform) a go, with the recently merged homing code.  Works perfectly (for me).  Excellent. :smile:

justinclift, 2018-07-15:
Ahhh, just found one weirdness.  Not sure if it's TinyG specific though.

With the "UGS Platform" nightly, in the "State" tab there is a drop down for Units.

I clicked on it (just to see what happens) and chose "Inches (G20)" - Note, I'm a mm guy, I was just experimenting.  The UI seems to hmm... get stuck?

"Get stuck" is probably not the correct term.  In the "Controller State (DRO)" tab the status changes from "IDLE" to "RUN", and never leaves it.  The fields in the State tab are greyed out meanwhile (eg permanently).

It's possible to click the Stop button in the toolbar, which seems to cancel the change properly.

Looking in the Console tab, a `G20` was clearly sent and accepted.  I'm guessing here, but maybe the code to handle the return status from g2core isn't in place (yet) for changes driven by the "State" tab?

breiler, 2018-07-16:
No you don't need to file a report. A fix is coming, it's just waiting on a code review. :smiley: 
Thanks for testing, soon you'll be able to use work coordinates as well.

justinclift, 2018-07-16:
Ahhh, and now I notice #1078 exists.  Cool. :smile:

justinclift, 2018-07-20:
> This is my "fault". I created a new plugin for jogging that would scale better in the UI and provide continuous jogging (you can find it under Window -> Plugins -> Jog controller). I didn't want to mess with the existing working module. 

Your new jogging plugin seems pretty good, and an improvement over the existing default one.

Is there any chance of getting that enabled by default for new users?  Preferably with the separate XY/Z step enabled too, but that step thing might just be me.  Not sure if people using GRBL commonly keep the step sizes the same or not. :smile:

justinclift, 2018-07-20:
And now I found a bug in it.  Can't seem to change the feed rate from 10k by typing a new number in.  eg Typing <kbd>5000</kbd> then pressing <kbd>enter</kbd>... and the number goes straight back to 10,000.

breiler, 2018-07-21:
Fixed the feed rate limit: #1088
Added issue to make the new jog module the default: #1089

breiler, 2018-07-25:
Closing this, start a new issue if problems occur.

closed 2018-07-25 by breiler