winder/Universal-G-Code-Sender #1077: "zed step size" not enbled

jiarko, 2018-07-11:
## Problem description 

 Hi, I'm trying to use the uGS platform on my old netbook equiped with windows 7 home 32 bit.

I've realized that both UGS platform and the std unoversal gcode sender have the "zed step size" spin box not enabled. If anyhow I send z movement command the machine behaves correctly with both software.

here a screenshot

[image]

any suggestion on how to fix this issue?

thanks
Andrea

winder, 2018-07-11:
Are you asking how to enable that setting? There is a “separate z step
size” setting in the program menu or settings menu.

jiarko, 2018-07-11:
Could you pls tell me where this setting is?

cause if I use another pc without touching anything (I mean just unzip the folder and start the program)  the "ZStep size" is enabled by default. that's why I've thought it is an issue.

thanks

breiler, 2018-07-11:
You should find it in the menu:
[image]

jiarko, 2018-07-11:
thanks, I'll check this setting!!
thanks!!!

jiarko, 2018-07-12:
the "issue2 is fixed as you suggested!!!

dont know why same program started oin 2 different machine has this setting not set in the same way

thanks a lot for the quick answer!!!

closed 2018-07-12 by jiarko