winder/Universal-G-Code-Sender #1079: Platform crashing after starting

olsarge03, 2018-07-12:
[hs_err_pid8928.log](https://github.com/winder/Universal-G-Code-Sender/files/2189490/hs_err_pid8928.log)

I have attached the error log. The classic build is working fine. Only the platform gui is crashing after starting. Previous platform nightly releases was working. I don't update often so the build that was working is probably from a month ago.

breiler, 2018-07-14:
We can start with upgrading your Java from version 1.8.0_141 which is about a year old.
You can find a new version here: https://java.com/en/download/manual.jsp

olsarge03, 2018-07-15:
Java said I was running the most current version. But I uninstalled and reinstalled Java. It is still crashing.
[image]
[hs_err_pid9428.log](https://github.com/winder/Universal-G-Code-Sender/files/2196005/hs_err_pid9428.log)

olsarge03, 2018-07-17:
new nightly, error log shows latest version of java, still crashing
[hs_err_pid12136.log](https://github.com/winder/Universal-G-Code-Sender/files/2202490/hs_err_pid12136.log)

zbuckholz, 2018-07-23:
I am having the same issue, just decided to download latest and check out what's new. 

It worked once, it showed me the new startup screen, I clicked the box telling it not to show again. Then closed the app, and reopened it. Now it just crashes as soon as it launches and shows the nightly build popup box.

[hs_err_pid14415.log](https://github.com/winder/Universal-G-Code-Sender/files/2220309/hs_err_pid14415.log)

breiler, 2020-10-02:
Not able to reproduce and no one else has reported similar issues. Closing.

closed 2020-10-02 by breiler