winder/Universal-G-Code-Sender #1082: Build error, cannot find symbol: getX(), getY(), getZ(), setX(), setY(), setZ()

gejza74, 2018-07-16:
Hi,

i try to run the build script, and i always get error by ugs-core:
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/model/Position.java:[95,24] cannot find symbol
  symbol:   method getX()
  location: class com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/model/Position.java:[97,24] cannot find symbol
  symbol:   method getY()
  location: class com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/model/Position.java:[99,24] cannot find symbol
  symbol:   method getZ()
  location: class com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGController.java:[281,44] cannot find symbol
  symbol:   method getZ()
  location: class com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGUtils.java:[140,26] cannot find symbol
  symbol:   method setX(double)
  location: variable workCoord of type com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGUtils.java:[144,26] cannot find symbol
  symbol:   method setY(double)
  location: variable workCoord of type com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGUtils.java:[148,26] cannot find symbol
  symbol:   method setZ(double)
  location: variable workCoord of type com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGUtils.java:[154,29] cannot find symbol
  symbol:   method setX(double)
  location: variable machineCoord of type com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGUtils.java:[158,29] cannot find symbol
  symbol:   method setY(double)
  location: variable machineCoord of type com.willwinder.universalgcodesender.model.Position
[ERROR] /Applications/CNC/Universal-G-Code-Sender/ugs-core/src/com/willwinder/universalgcodesender/TinyGUtils.java:[162,29] cannot find symbol
  symbol:   method setZ(double)
  location: variable machineCoord of type com.willwinder.universalgcodesender.model.Position
[INFO] 10 errors 
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] ugs-parent 2.0-SNAPSHOT ............................ SUCCESS [  0.660 s]
[INFO] ugs-core ........................................... FAILURE [  5.685 s]
[INFO] ugs-platform-parent ................................ SKIPPED
[INFO] ugs-platform-branding .............................. SKIPPED
[INFO] ugs-platform-ugslib ................................ SKIPPED
[INFO] ugs-platform-ugscore ............................... SKIPPED
[INFO] ugs-platform-visualizer ............................ SKIPPED
[INFO] ugs-platform-plugin-workflow ....................... SKIPPED
[INFO] ugs-platform-gcode-editor .......................... SKIPPED
[INFO] ugs-platform-surfacescanner ........................ SKIPPED
[INFO] ProbeModule ........................................ SKIPPED
[INFO] DowelModule ........................................ SKIPPED
[INFO] GcodeTools ......................................... SKIPPED
[INFO] ugs-platform-plugin-jog ............................ SKIPPED
[INFO] ugs-platform-plugin-setup-wizard ................... SKIPPED
[INFO] ugs-platform-welcome-page .......................... SKIPPED
[INFO] ugs-platform-app 2.0-SNAPSHOT ...................... SKIPPED
[INFO] ------------------------------------------------------------------------

Anybody knows what's wrong here?

I try it on: OSX 10.11.6, Java 8, Maven 3.5.4

Thank you

breiler, 2018-07-16:
Might be an old compilation of Position-class, have you tried running `mvn clean install`?

winder, 2018-07-16:
I believe this is caused by OSX coming pre-installed with an ancient version of `vecmath.jar`, I'm not sure why maven doesn't override it. I'm working on a PR for >3 axes and removed `vecmath.jar` completely.... but that won't help today.

winder, 2020-04-09:
This is an old known issue, it's probably still a problem for old mac computers but there are no plans to address it directly.

closed 2020-04-09 by winder