winder/Universal-G-Code-Sender #1090: "Reset Zero" & "Return to Zero" - separate out Z axis reset/return

justinclift, 2018-07-21:
## Feature request

When setting up a job on my CNC machine, it's fairly common to set the X and Y co-ordinates by jogging the end mill tip to the correct location then reseting the machine co-ordinates for those axis.

With the Z axis though... the current Reset/Return buttons affect that too.  So it's not possible to use these buttons effectively without weird contortions as (for me) the X0 Y0 Z0 position is commonly at a point inside the material.  eg a spot I really don't want to have to drill into the material to reach, just in order to set the Z0 point there.

With other G-Code senders, the approach taken is to allow zero-ing of each axis separately, and the "return to zero" moves to the Z/Y zero but leaves the Z a specific distance above the Z0 (6mm in the case of Carbide Motion), I don't remember what CNCjs does off hand.

Is it feasible to have the Reset/Return bits separate out things instead?  At least for the Z axis, though having the X and Y separate as well would be useful for some people. :smile:

breiler, 2018-07-21:
Resetting individual axes is possible using the "axis reset buttons":
(from the user guide: http://winder.github.io/ugs_website/guide/platform/#controller-state-dro)
[image]

For invidual buttons to return to zero could be solved using macros. Here showing the macro configuration dialog and the macros panel:
[image]

We could add buttons, but space is limited. Do you have any ideas where to put such functions?

justinclift, 2018-07-21:
Thanks @breiler. :smile:

Let me experiment with the DRO buttons and macros tomorrow, and I'll see if there's still a need for extra buttons.  Probably should look through the User Guide while I'm at it too. :smile:

justinclift, 2018-07-25:
DRO buttons worked exactly as you mentioned.  Haven't tried the macros yet. :angel:

breiler, 2019-02-19:
It looks like this is resolved?

closed 2019-02-19 by breiler

justinclift, 2019-02-19:
Oops, completely forgot about this.  Yep, the DRO buttons did the job so I didn't come across a need to do more stuff with it. :smile: