winder/Universal-G-Code-Sender #1094: UGS Not working on Rasperry Pi 3

jratha, 2018-07-26:
trying to run UGS classic and platform on RPI3.

install goes well but no splash scren and no UGS.

> Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-jdesktop-layout/RELEASE82/org-jdesktop-layout-RELEASE82.pom (614 B at 1.                           5 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-api-htmlui/RELEASE82/org-netbeans-api-htmlui-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-api-htmlui/RELEASE82/org-netbeans-api-htmlui-RELEASE82.pom (3 K                           B at 7.1 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-api-progress-compat8/RELEASE82/org-netbeans-api-progress-compa                           t8-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-api-progress-compat8/RELEASE82/org-netbeans-api-progress-compat                           8-RELEASE82.pom (2 KB at 2.5 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-api-search/RELEASE82/org-netbeans-api-search-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-api-search/RELEASE82/org-netbeans-api-search-RELEASE82.pom (4 K                           B at 8.4 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-api-visual/RELEASE82/org-netbeans-api-visual-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-api-visual/RELEASE82/org-netbeans-api-visual-RELEASE82.pom (2 K                           B at 2.7 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-execution/RELEASE82/org-netbeans-core-execution-RELEA                           SE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-execution/RELEASE82/org-netbeans-core-execution-RELEAS                           E82.pom (3 KB at 3.5 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-io-ui/RELEASE82/org-netbeans-core-io-ui-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-io-ui/RELEASE82/org-netbeans-core-io-ui-RELEASE82.pom                            (2 KB at 3.2 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-core-multitabs/RELEASE82/org-netbeans-core-multitabs-RELEASE82                           .pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-core-multitabs/RELEASE82/org-netbeans-core-multitabs-RELEASE82.                           pom (3 KB at 5.7 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-nativeaccess/RELEASE82/org-netbeans-core-nativeaccess                           -RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-nativeaccess/RELEASE82/org-netbeans-core-nativeaccess-                           RELEASE82.pom (2 KB at 3.4 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-core-netigso/RELEASE82/org-netbeans-core-netigso-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-core-netigso/RELEASE82/org-netbeans-core-netigso-RELEASE82.pom                            (2 KB at 3.7 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-network/RELEASE82/org-netbeans-core-network-RELEASE82                           .pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-network/RELEASE82/org-netbeans-core-network-RELEASE82.                           pom (2 KB at 4.3 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-osgi/RELEASE82/org-netbeans-core-osgi-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-osgi/RELEASE82/org-netbeans-core-osgi-RELEASE82.pom (2                            KB at 3.6 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-output2/RELEASE82/org-netbeans-core-output2-RELEASE82                           .pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-output2/RELEASE82/org-netbeans-core-output2-RELEASE82.                           pom (3 KB at 5.5 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-ui/RELEASE82/org-netbeans-core-ui-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-ui/RELEASE82/org-netbeans-core-ui-RELEASE82.pom (4 KB                            at 7.6 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-core-windows/RELEASE82/org-netbeans-core-windows-RELEASE82                           .pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-core-windows/RELEASE82/org-netbeans-core-windows-RELEASE82.                           pom (4 KB at 7.4 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /external/org-netbeans-html-ko4j/RELEASE82/org-netbeans-html-ko4j-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           external/org-netbeans-html-ko4j/RELEASE82/org-netbeans-html-ko4j-RELEASE82.pom (                           420 B at 0.7 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /external/org-netbeans-html-xhr4j/RELEASE82/org-netbeans-html-xhr4j-RELEASE82.po                           m
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           external/org-netbeans-html-xhr4j/RELEASE82/org-netbeans-html-xhr4j-RELEASE82.pom                            (421 B at 0.9 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-lib-uihandler/RELEASE82/org-netbeans-lib-uihandler-RELEASE82.p                           om
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           api/org-netbeans-lib-uihandler/RELEASE82/org-netbeans-lib-uihandler-RELEASE82.po                           m (944 B at 2.1 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /modules/org-netbeans-libs-felix/RELEASE82/org-netbeans-libs-felix-RELEASE82.pom
Downloaded: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans/                           modules/org-netbeans-libs-felix/RELEASE82/org-netbeans-libs-felix-RELEASE82.pom                            (782 B at 1.7 KB/sec)
Downloading: http://bits.netbeans.org/nexus/content/groups/netbeans/org/netbeans                           /api/org-netbeans-libs-javafx/RELEASE82/org-netbeans-libs-javafx-RELEASE82.pom
[... cut ...]
> [ERROR] No X11 DISPLAY variable was set, but this program performed an operati                           on which requires it.
> [ERROR] -> [Help 1]
> [ERROR]
> [ERROR] To see the full stack trace of the errors, re-run Maven with the -e sw                           itch.
> [ERROR] Re-run Maven using the -X switch to enable full debug logging.
> [ERROR]
> [ERROR] For more information about the errors and possible solutions, please r                           ead the following articles:
> [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutio                           nException
>

breiler, 2018-07-26:
Are you SSH:ing to the RPi when trying to compile or are you running it with an attached screen?

jratha, 2018-07-26:
Using ssh to compile but the intention is to use a 7 inch screen when it arrives

carneeki, 2018-07-26:
Are you also using ssh X forwarding to get the GUI back to your main
computer? If so, double check that your ssh client is forwarding X ports
correctly (I think it's ssh -X destination).

I had problems with X forwarding over SSH (mostly latency problems despite
being on local connection) but using NoMachine as a remote desktop works
well for me and it seemed to solve the latency issue, and I think it might
be how due to the graphical environment is created on the Pi. I'd recommend
try running UGS through a remote desktop utility (such as NoMachine/Remote
Desktop).

jratha, 2018-07-26:
Thanx @carneeki  

Couldn’t get realvpn or Remote Desktop to work, but dug out old keyboard and mouse and entered terminal directly on pi, now running. Pity visualisation doesn’t work, but I can live without that

closed 2018-07-26 by jratha