winder/Universal-G-Code-Sender #1101: cannot be run from folder that contains non-ASCII characters in path

sua88, 2018-07-31:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

I can't start ugs platform on my second computer, I have got this message : 
cannot be run from folder that contains non-ASCII characters in path 

it's working pretty good on my another one ( I did the instalation at the the same time)

I dont know what to do!
please help

sua88, 2018-07-31:
thanks breiler, I just updated my first post;

breiler, 2018-07-31:
A similar issue exists on MacOSX and probably Linux as well. The problem is probably the `é` in `Propriétaire`. Place it in another directory or rename that directory and it should work.

sua88, 2018-08-01:
thanks breiler it's working like a charme now 
thanks for the help

closed 2018-08-01 by breiler

DmitryEvdochenko, 2019-03-09:
была такая же ошибка. Изначально разархивировал в директорию, где по пути были папки, в названии которых содержались символы (!!). После того, как переместил в другое место все заработало