winder/Universal-G-Code-Sender #1108: Bit crash at end of job - no retraction.

Beardy241, 2018-08-10:
Three times today, with three different files, I experienced a marauding bit crashing across my work.

Fusion360 > UGS Platform > Ox CNC.

Instead of retracting the bit at the end of the job then returning to a home position, the bits stays at depth and is dragged directly home — kicking and screaming — gouging a path as it goes.  It happens at 'rapid' speed so it's all rather exciting.

I copied the following output from the end of the Console window when I noticed it said "Skipping command"…

`
>>> X694.02Y429.8535Z-17.856
ok
>>> X694.0125Y429.8165Z-17.7647
ok
>>> X694.008Y429.7938Z-17.6686
ok
>>> X694.0065Y429.7862Z-17.57
>>> G0Z15
Skipping command #78621
>>> G53Z5
>>> M5
ok
>>> G4P1.0
>>> G53X0Y900
Skipping command #78626
Skipping command #78627
ok
ok
ok
ok
ok
ok
ok

**** Finished sending file in 00:44:47 ****

and…

>>> X694.008Y429.7938Z-17.6686
ok
>>> X694.0065Y429.7862Z-17.57
>>> G0Z15
Skipping command #85537
>>> G53Z10
>>> M5
ok
>>> G4P1.0
>>> G53X0Y900
Skipping command #85542
Skipping command #85543
ok
ok
ok
ok
ok
ok
ok

**** Finished sending file in 00:25:54 ****

and…

>>> X380.6344Y29.5665Z-34.5102
ok
>>> X380.6497Y29.5627Z-34.37
>>> G0Z6
Skipping command #26691
>>> G53Z5
>>> M5
ok
>>> G4P1.0
>>> G53X0Y222
Skipping command #26696
Skipping command #26697
ok
ok
ok
ok
ok
ok
ok

**** Finished sending file in 00:47:10 ****
`

Most strangely, I noticed that it was crashing to a home position for a PREVIOUS job!

### Specifications
#### Version 
  UGS Platform 2.0 - [nightly] Jul 23, 2017

#### Operating system
  Mac OSX

#### Platform
  GRBL 1.1f

breiler, 2018-08-10:
Sorry to hear this!

This could be a result of the error handler:
http://winder.github.io/ugs_website/guide/common/#error-handling

The version of UGS from Jul 23, 2017 is pretty old and the problems you are experiencing might have been fixed. The problem with commands running from previous jobs could be a result from buffers not being cleared which we have worked with earlier this year.

Do you mind sharing the files you tried to run?

Beardy241, 2018-08-10:
Thanks for that Joacim.

I've now updated the UGS software.  (I had previously done a 'Check for Updates' which had come back empty).

One file attached.

[tree right final pass.nc.zip](https://github.com/winder/Universal-G-Code-Sender/files/2277572/tree.right.final.pass.nc.zip)

breiler, 2018-08-10:
I've run the file and got this in the end:
```
>>> X694.0125Y429.8165Z-17.7647
ok
>>> X694.008Y429.7938Z-17.6686
ok
>>> X694.0065Y429.7862Z-17.57
>>> G0Z15
Skipping blank line #5272
>>> G53Z10
>>> M5
ok
>>> G4P1.0
>>> G53X0Y900
Skipping line: M30
Skipping comment-only line: (%)
ok
ok
ok
ok
ok
ok
ok

**** Finished sending file in 00:11:42 ****
ok
```

The skipped commands didn't seem to matter. However the G53Z10 was a bit dangerous as it moves to the machine coordinates Z10. This is probably fine for your machine but on my it was totally unexpected (it moved to work position Z70.5 which was safe in my case) and I didn't know what was going on. The visualiser don't show this either so it could probably be improved.

zbuckholz, 2018-09-16:
I am having an issue that might be similar. My job runs fine, at the end I click return to zero and it moves x and y properly, then z goes down past zero and crashes into the vise. I was able to reproduce the problem two times. Initially I thought it was backlash issues, but not so sure about that now.
I have attached the gcode file and a photo showing the end result. 
[1001.txt](https://github.com/winder/Universal-G-Code-Sender/files/2386229/1001.txt)

[image]

winder, 2018-09-18:
Sorry to hear about the crash, and damage done to your tools. What version of UGS are you using? I removed this "feature" a long time ago so I'm hoping you are using an older version of UGS. If you are, please give the latest nightly build a try and "return to zero" should be better.

stanhowe, 2019-01-25:
I had a simular "skip command" but it was in the middle of a run whee the bit was supposed to raise to safe height, travel diagonally. then plunge again. Instead it stayed at the lower height and plowed across the project (fortunately in an area to be scrap) I hit my E-stop. Upon looking at the command window I found it had "Skip command xx" in both of the z axis steps. Verified against the g-code and the steps looked as I expected. I'm unclear if the "skip command" comes from the UGS or from the firmware on my machine?  I plan to re-run same code using UGS and another code sender to see if I can recreate the issue on either or both. attached is photo of the crash, screen shot of the tool path in CAM software, screenshot of command window with skip commands, screen shot of the G-code at the same spot as the skipped commands.
[image]
[image]
[image]
[image]

breiler, 2019-03-02:
@stanhowe sorry for the late reply! The "skip command" comes from UGS. Which version of UGS were you running? Do you have the gcode so we can try to reproduce the problem?

breiler, 2019-03-02:
Adding the bug label to the issue because of the G53Z10 command that doesn't show properly in the visualizer. I'm not sure if it will work but the visualizer should display this job using the G53 offsets.

stanhowe, 2019-03-02:
Attached is a photo of my crash, it was supposed to lift move to the lower right side and drop down again.it ignored both of the z axis moves,  also attached is the g-code I was running. I believe I was running the July23rd 2017 version because the touch pad wasn’t working in the December 2018 version.

I believe I uploaded screen shots to github of the skip commands, and the on screws visualizer. I just deleted them from my desktop 2 days ago when i was cleaning up my desktop.