winder/Universal-G-Code-Sender #1109: No Serial Ports found ?

CCTofol, 2018-08-14:
## Problem description 

### Expected Behavior

### Actual Behavior

Then suddenly I was able to connect.
Started configuring grbl for a few hours.
Then lost connection and haven't been able to get it back, comp, cables, Arduino, all same as when connection made.

### Steps to Reproduce the Problem
  1. Open the program
 No serial ports found
2. click refresh
No serial ports found

### Specifications
#### Version

#### Operating system

#### Platform

breiler, 2018-08-15:
Is it possible for you to try out a later version of GRBL, the build you are using seems pretty old?

CCTofol, 2018-08-15:
I thought that was the one I downloaded from https://github.com/gnea/grbl yesterday.
Anyway, downloaded again replaced 'old' version in Arduino library.
Opened Grblupload.
Compiles ok.
But when I upload It stalls at around 90% of write and says "avrdude: stk500_recv(): programmer is not responding" repeatedly.

Tried uploading a different sketch and it uploaded fine, so my guess is Arduino is ok.
Anything else I should try?

breiler, 2018-08-15:
You could try fetching a binary version and try uploading it using HexUploader:
https://github.com/grbl/grbl/wiki/Flashing-Grbl-to-an-Arduino

CCTofol, 2018-08-15:
Thanks breiler, some degree of success from your suggestion.
Managed to upload grbl using Hexuploader.
UGS now sees a usb port in the ports window. 
when I click 'open' the console reports
**** Connected to /dev/tty.usbmodem1D11141 @ 115200 baud ****
The jog buttons etc become active but do not operate the machine.

With machine powered off and usb disconnected I get the exact same behaviour as above.

CCTofol, 2018-08-15:
This morning I'm back to the beginning. No ports found.

breiler, 2018-08-16:
Is the Arduino a genuine or a cheaper China version?

CCTofol, 2018-08-16:
I just got another Arduino Uno (knock off I think) and it works.
Phew.
The other two I was trying where a Freetronics eleven and another generic knock off.
Both work fine with other projects and play nice with Arduino UDI which is what was frustrating me.
Nothing pointed to the Arduinos in my mind

Got my fingers crossed this one continues to work.
Thank you for you help and your patience.

breiler, 2018-08-17:
Is everything working as expected now, can we close this issue?

CCTofol, 2018-08-17:
Yes, thank you Joacim, all is working fine at the moment.

P +64 21 724 943
e chris@chriscameron.co.nz
s cctofol
u chriscameron.nz

closed 2018-08-18 by breiler