winder/Universal-G-Code-Sender #1110: UGS Plataform - Autoleveler

ALL1991, 2018-08-15:
Good afternoon,
I have a problem with Autoleveler, when I send it to scan it performs perfectly, when I create generate test data and apply to gcode, the original speed parameters of the gcode are changed to a very slow speed, being not feasible to use, the original and generated code in Artcam.
If anyone can help, I appreciate it.

ALL1991, 2018-08-15:
(error:22) Feed rate has not yet been set or is undefined.(error:22) Feed rate has not yet been set or is undefined.

alecarvalhof, 2021-04-15:
Hello friends.
I am desperately looking for the solution to this same problem that is making me no longer use the UGS that I like so much.
Someone could provide a solution.
@ ALL1991 already managed to solve this problem? if so please inform.
I continue with hope!

breiler, 2024-12-03:
This should now be fixed, it should respect the units of the original Gcode, please try the latest version of UGS.

closed 2024-12-03 by breiler