winder/Universal-G-Code-Sender #1111: Setup wizzard does not support board

CW3Davis, 2018-08-22:
# Problem description 
I can connect to my tinyg but it will not control it. I tried using the setup wizard. After it connects I get a message stating wizard does not support my board. 

### Specifications
#### Version

  What versions of the application does the issue exist in.  You find the version 
  in the title bar of the application. Example:
  
  UGS Platform 2.0 - Nightly Build Mar 23, 2018
  UGS Classic 1.0.9
-->both

#### Operating system

#### Platform

breiler, 2018-08-22:
The build from March 23 is too old. Please upgrade to the latest nightly build: https://github.com/winder/Universal-G-Code-Sender#downloads

The setup wizard does not yet support TinyG so you need to configure manually using the "Firmware settings"

CW3Davis, 2018-08-22:
Not exactly sure how to do the firmware setup. Is this a download file? Can you point me in the right direction? Thanks

breiler, 2018-08-22:
Just download the latest version of UGS: https://github.com/winder/Universal-G-Code-Sender#downloads

For configuring TinyG you can use the Firmware Settings from the menu to set the parameters:
[image]

[image]

CW3Davis, 2018-08-23:
I had a chance to check the firmware. When I clicked on it I got a message now firmware found.

breiler, 2018-08-23:
So it's ok to close this issue?

CW3Davis, 2018-08-23:
No I cannot configure the Universal Gcode Sender to work with my tinyg board. I have downloaded the most recent copy of UGSplatform. I tried to use the firmware to configure it as suggested, but I get a message "No firmeware found".

winder, 2018-08-23:
Is this your first time setting up this board? Are you able to connect to your TinyG with something like Coolterm according to the directions here:
https://github.com/synthetos/TinyG/wiki/Connecting-TinyG#establish-usb-connection

CW3Davis, 2018-08-23:
I have been using the board for about a year. I usually use Chillipeper. The machine works perfectly, but I don’t like being dependent on the internet connection. I live in an area with crappy internet.

momiccioli, 2019-06-17:
Has this been resolved?  I am having the same issue, yes the TinyG is working fine and accessible.  UGS will do some basic stuff but Jog controls are greyed out.  The Home button works.  I am using a nightly build from April 23, 2019.  Here is the version info of the TinyG

tinyg [mm] ok>
[fb]  firmware build            440.20
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00
[id]  TinyG ID                    3X3566-EUZ

Would love to see this work with Autoleveling

breiler, 2019-06-17:
@momiccioli  Could you post the console message from when you first connect to the contromler?

momiccioli, 2019-06-17:
Here is a screen shot when I fire up UGS:
[image]

Here is a screen shot when I connect to the TinyG

[image]
The only message in the console area is:
`**** Connected to COM3 @ 115200 baud ****`
Which is correct for the TinyG

JacksCrafts, 2019-06-17:
Are you on the right Com Port?

momiccioli, 2019-06-17:
Yes I am it can be accessed by other software and some very basic functions with UGS work.  I can send commands via the command line in the console area.  The home button works.  None of the jog buttons work though.  And I can't manually code firmware settings.  Is there a specific version of the TinyG that is supported?

JacksCrafts, 2019-06-17:
[image]

I am using Arduino. But when I connect it automatically shows my settings.

momiccioli, 2019-06-17:
When I test with an Arduino I get the same result you do.   It clearly understands the Arduino.  I can connect to the TinyG but when I go the wizard I get the following:

[image]

Its as if it does not what to do with the TinyG

breiler, 2019-06-17:
I suspect that this has to do with the initialization of the TinyG-board which behaves differently than g2core (which is the platform I've developed against).

When connecting to the controller UGS expects a message from TinyG (a "r"-response with the message "SYSTEM READY"). But it's not shown in the screenshots that you've provided which is why some functions isn't working. When such a message is retrieved from the controller and additional setup is done by UGS.

You could try hit the button "Soft reset" which will send a reset command to TinyG and reinitialize the board.

breiler, 2019-06-17:
Oh, and the setup wizard doesn't not yet work with TinyG/g2core boards. You'll see the board version, but that's about it. =(

momiccioli, 2019-06-17:
I saw that the setup wizard is not working yet.  If I do a soft reset it does start talking to the TinyG a little better.  I can live without the Firmware setup since I can do that via terminal.  There are some other buttons that don't appear to work like soft set to zero and I am thinking because UGS does not understand the responses.  Is that because I have an older TinyG.  How hard would it be to include support for older TinyGs if that is the issue.  I would be willing to help by testing maybe coding etc depending on the complexity.

breiler, 2019-06-17:
I don't think it will be that hard if you got the hardware, which I don't. =(

The g2core/tinyg implementations should probably be splitted apart so the differences in the protocols could be handled. 

The current controller implementation is found here:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/TinyGController.java

This is the part of the initialization that didn't work on your hardware:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/TinyGController.java#L181

momiccioli, 2019-06-17:
Let me see if I can interpret this, been a while since I coded.

norru, 2019-09-01:
@breiler I am using UGS with g2core and I get the same results.

norru, 2019-09-01:
> When such a message is retrieved from the controller, additional setup is done by UGS.

I don't see this message in the terminal either.

norru, 2019-09-01:
@breiller Let me know if you want anything tested.

breiler, 2019-11-02:
@norru and @CW3Davis I would really appreciate if you had the time to try both TinyG and g2core. They now have different firmware selections in the drop down box.

I've seen some smaller issues eg. when switching between different unit modes (g20/g21), where the DRO gets confused. The M0-command will also cause some weird behaviour creating offsets in the visualiser. Once people start using it we will probably find more stuff...

breiler, 2020-03-20:
Please create a new issue if this still is a problem.

closed 2020-03-20 by breiler

doanhung2691, 2022-01-27:
[image]
[image]
I'm operating a cnc pen plotter, but it's not working, so you guys can help me? how can I fix it?
I use UGS 2.9 and win 10
Thanks

breiler, 2022-01-27:
@doanhung2691 please create a new issue and provide details about your controller board.