winder/Universal-G-Code-Sender #1114: UGS Platform - Visualizer upside down

ione2k, 2018-08-26:
Hi all,

I was wondering if I can get a little help.

I recently started using UGS Platform to control my CNC machine. For the most part it has been working great.

I just have a small issue with Visualizer. I run UGS on MAC OS to see what the G-Code will do after exporting the file from the CAM software and it works great.

Then, I copy the g-code file to a tablet with windows 8 which drives the CNC machine.

I noticed that the same version of UGS Platform has the visualiser “upside down” on Windows. I know that there is no axis configuration problem because I can see a small writing in white “FPS X.XX” at the BOTTOM-Right of the Visualizer on the MAC OS version and the Windows 8 version has the same writing with the fonts “upside down” at the TOP-Right corner. The machine cuts the way it should be but I see tool location icon also upside down pointing up instead of down in visualizer. There is also no “orientation cube” even though the option is selected.

I have tried the stable and many nightly build versions and they all behave the same way. I am wondering if it has to do with the tablet having a discrete video card?

Thank you in advance for your help.

winder, 2018-08-27:
That's really weird, could you share a screenshot? I'm really not sure what this could be, maybe try updating the graphics driver?

breiler, 2019-08-12:
This looks like some graphics card issue and there's not much we can do with it... =( 
I'm closing this.

closed 2019-08-12 by breiler