winder/Universal-G-Code-Sender #1116: Problems with Universal Gcode Sender - Error while processing resoponse

cncvictor, 2018-09-02:
Hello everyone, I have a problem, when I connect the Universal Gcode Sender it shows a message and it keeps appearing this message without stopping.

I need help, please.

Victor
[image]
[image]

cncvictor, 2018-09-02:
I have a mini CNC, those with pen, I'm studying the whole mechanism to be able to make a bigger one. Take a look on my youtube channel that you will see my machine when it was working up to 1 week ago
https://youtu.be/xUxPWlC78ZE

breiler, 2019-02-19:
A lot has been changed with the processing of the responses in UGS, please upgrade to the latest nightly build.

closed 2019-02-19 by breiler