winder/Universal-G-Code-Sender #1118: Arc interpolation problem in the ZX plane

RafaelEKH, 2018-09-03:
Hi!
I'm new to github, so please bear with me.

## Problem description 
Arc interpolation offset mode not working correctly
### Expected Behavior
G18 is the selected plane.  
G01X20Z20 - go to the starting point
G02 X60 Z40 K20 I15 where K is offset from Z and I is offset from X starting point

[image]
Expected behavior, but the picture is XY-plane, with G17 and corresponding X Y coordinates

### Actual Behavior
same gcode, and the actual result is:
[image]

I tried leaving it as I and J and that gave this result:
[image]

### Steps to Reproduce the Problem
Just open the program. Send the written G-codes, dont forget about the plane selection G17/G18

#### Version
Universal Gcode Sender Version 2.0 Aug. 26 nightly

#### Operating system
Windows 10 64 bit

#### Platform
  GRBL 1.1f

#### Other
I've tested it from the ardunio IDE  serial monitor and there the code works as expected.

winder, 2018-09-04:
Does your gcode file specify G17/G18? They should be supported, there is a test file here: https://github.com/winder/Universal-G-Code-Sender/blob/master/test_files/g17-g18-g19.gcode

RafaelEKH, 2018-09-04:
Okay, I double checked it.

Your example is in the radius mode with R, my problem is in the center offset mode.
I modified your code to center offset and it works!
`F250.
G90 G20

G17
G00 X-1.5 Y0.
G02 X1.5 Y0 I1.5 J0
G02 X-1.5 Y0 I-1.5 J0

G18
G00 X-1.5 Y0.
G02 X1.5 Z0. I1.5 K0
G02 X-1.5 Z0. I-1.5 K0

G19
G00 X0 Z0 Y-1.5
G02 Z0 Y1.5 J1.5 K0
G02 Z0 Y-1.5 J-1.5 K0
`
[image]

Now here is my code:
`G20 G90
G18 
G01 Z20 X20 Y0
G02 Z40 X60 I15 K20

G17
G00 X0 Y0 Z0
G01 X20 Y20 
G02 X40 Y60 J15 I20
`

and the result is:
[image]

So my guess was something might be with your K parameter being 0, to check this let's try this code:
`F250.
G90 G20

G18
G00 X0 Y0 Z0
G00 Z1.5 X0
G02 Z-1.5 X0 I0 K-1.5
G02 Z1.5 X0 I0 K1.5`

still working as expected:
[image]

I don't know where is the fault in my code, but it was giving anomalies with exported g-codes as well not just with my hand written example. The working examples always using a 180 degree arc, maybe it has a problem with other sections like in my code?

dastultz, 2018-09-05:
On Mon, Sep 3, 2018 at 12:31 PM RafaelEKH <notifications@github.com> wrote:

> Hi!
> I'm new to github, so please bear with me.
> Problem description
>
> Arc interpolation offset mode not working correctly
>
Perhaps this is related to this issue:
https://github.com/winder/Universal-G-Code-Sender/issues/853

/Daryl

>

RafaelEKH, 2018-09-05:
dastultz
It is probably related. I read your thread and after I tested the my first example code and also my yesterday's but with G03 and G21, and despite the preview, it does what it should do. However I don't know what is the arc expander.

[image]

After thinking about it, it's logical it does what it should as it's up to grbl to execute the G-code not UGS.
Should I change the title to something like "Arc rendering problem outside of the XY plane" ? I'm not sure if the YZ is affected though. 

Still it's really interesting that sometimes it renders as expected and sometimes not....

DanielWeigl, 2019-04-29:
this is most likely fixed in https://github.com/winder/Universal-G-Code-Sender/pull/1220

closed 2019-04-29 by winder