winder/Universal-G-Code-Sender #1119: grbl has not finished loading

lorcanwall, 2018-09-04:
when i load and prepare my cnc it will come up with an error message saying grbl has not finished booting. this is very annoying as i need to get my cnc working.

winder, 2018-09-10:
This is generally a configuration issue, some details here:
http://winder.github.io/ugs_website/guide/troubleshooting/#grbl-has-not-finished-booting

steve2718, 2018-10-12:
Hi Will, Same issue. HW: RPi 3 / Protoneer hat. GRBL is flashed onto P-board out of the box. UGSplatform2.0 stable (beta) runs fine, but the nightly build of platform generates the "grbl not finished booting" error. Baud = 115200, Port: /dev/ttyAMA0 (< zero).

winder, 2018-10-12:
@steve2718 thanks for the report, could you attach the  `messages.log` file?

http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

steve2718, 2018-10-12:
Problem solved for me - UGSplatform latest / nightly build works on RPi and Protoneer GRBL board.
The Protoneer Rpi image comes with some sort of UGS (classic) as part of it. Long story short, I removed the ugs and ugsplatform directories and the .ugs & .ugsplatform directories. Re-installed. Works like a charm - operator error. But, that might be a 5th(?) bullet point to consider for troubleshooting notes regarding "grbl hasn't finished booting" - rm not only the main dirs, but also the hidden dirs and re-install.

breiler, 2019-03-02:
Added new trouble shooting option

closed 2019-03-02 by breiler