winder/Universal-G-Code-Sender #1120: Nightly Build Visualizer Not warking

Baybeyya, 2018-09-10:
My old UGS stopped showing visualizer. I removed old one and downloaded new nightly build UGS but still the visualizer is not working. 
I am on Windows 10. 
Appreciate suggestions. Thanks.

winder, 2018-09-10:
Could you share the log file? You can find it's location here:
http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

Baybeyya, 2018-09-10:
I sorry I am not very familiar but trying hard. I cannot find UGS logs in
the location. Here is screen shot.
c:\user\AppData\Roaming
Thanks

winder, 2018-09-10:
The directory is also listed in the about page while UGS is running.

Baybeyya, 2018-09-10:
Sorry. I could not find. will there be help guide for windows 10?

winder, 2018-09-10:
No, my guess is that it's the same as in Windows 7

Baybeyya, 2018-09-10:
{
  "Name": "GRBL",
  "Version": 4,
  "Controller": {
    "name": "GRBL"
  },
  "GcodeProcessors": {
    "Front": [
      {
        "name": "CommentProcessor",
        "enabled": true,
        "optional": false
      },
      {
        "name": "FeedOverrideProcessor",
        "enabled": false,
        "optional": true,
        "args": {
          "speedOverridePercent": 60
        }
      },
      {
        "name": "M30Processor",
        "enabled": true,
        "optional": false
      },
      {
        "name": "ArcExpander",
        "enabled": false,
        "optional": true,
        "args": {
          "segmentLengthMM": 1.3
        }
      }
    ],
    "Custom": [],
    "End": [
      {
        "name": "DecimalProcessor",
        "enabled": true,
        "optional": true,
        "args": {
          "decimals": 4
        }
      },
      {
        "name": "WhitespaceProcessor",
        "enabled": true,
        "optional": true
      },
      {
        "name": "M3Dweller",
        "enabled": false,
        "optional": tr

Baybeyya, 2018-09-10:
Now I get it. I am using classic version so I have to give Jason file. 
{
  "firmwareVersion": "GRBL",
  "fileName": "C:\\Users\\user\\Desktop\\Test2.nc",
  "port": "COM3",
  "portRate": "115200",
  "manualModeEnabled": false,
  "manualModeStepSize": 1.0,
  "useZStepSize": true,
  "zJogStepSize": 1.0,
  "jogFeedRate": 51.0,
  "scrollWindowEnabled": true,
  "verboseOutputEnabled": false,
  "commandTableEnabled": true,
  "mainWindowSettings": {
    "xLocation": -17,
    "yLocation": 12,
    "width": 1382,
    "height": 754
  },
  "visualizerWindowSettings": {
    "xLocation": -32000,
    "yLocation": -32000,
    "width": 160,
    "height": 39
  },
  "singleStepMode": false,
  "statusUpdatesEnabled": true,
  "statusUpdateRate": 200,
  "displayStateColor": true,
  "defaultUnits": "mm",
  "probeAxis": "Z",
  "probeFeed": 5.0,
  "probeDistance": -10.0,
  "probeOffset": 0.0,
  "showNightlyWarning": true,
  "showSerialPortWarning": true,
  "autoConnect": false,
  "autoReconnect": false,
  "toolbarIconSize": 0,
  "autoLevelSettings": {
    "autoLevelProbeZeroHeight": 0.0,
    "autoLevelProbeOffset": {
      "units": "UNKNOWN",
      "x": 0.0,
      "y": 0.0,
      "z": 0.0
    },
    "autoLevelArcSliceLength": 0.01,
    "stepResolution": 10.0,
    "probeSpeed": 10.0,
    "zSurface": 0.0
  },
  "fileStats": {
    "minCoordinate": {
      "units": "MM",
      "x": 0.0,
      "y": 0.0,
      "z": -1.0
    },
    "maxCoordinate": {
      "units": "MM",
      "x": 59.60659898477157,
      "y": 58.91878172588832,
      "z": 15.0
    },
    "numCommands": 34
  },
  "macros": {
    "1": {
      "gcode": "G91 X0 Y0;"
    }
  },
  "language": "en_US",
  "pendantConfig": {
    "stepSizeList": [
      {
        "value": ".1",
        "label": ".1",
        "selected": false
      },
      {
        "value": "1",
        "label": "1",
        "selected": false
      },
      {
        "value": "5",
        "label": "5",
        "selected": false
      },
      {
        "value": "10",
        "label": "10",
        "selected": true
      },
      {
        "value": "50",
        "label": "50",
        "selected": false
      }
    ],
    "shortCutButtonList": [
      {
        "label": "Return to Workpiece 0",
        "gCodeCommandList": [
          "G90 G0 X0 Y0 Z0"
        ]
      },
      {
        "label": "Start Spindle",
        "gCodeCommandList": [
          "M3"
        ]
      },
      {
        "label": "Stop Spindle",
        "gCodeCommandList": [
          "M5"
        ]
      },
      {
        "label": "Coolant On",
        "gCodeCommandList": [
          "M7"
        ]
      },
      {
        "label": "Coolant Off",
        "gCodeCommandList": [
          "M9"
        ]
      }
    ]
  }
}

winder, 2018-09-10:
@Baybeyya you'll have to run from the command line to get the error log in that case

winder, 2018-09-10:
Maybe this will help you with that: https://www.youtube.com/watch?v=MhwX8_ohOD8

Baybeyya, 2018-09-10:
Here is screen shot from cmd prompt

winder, 2018-09-10:
You'll have to share from github, your screenshots aren't showing up.

Baybeyya, 2018-09-10:
[image]

winder, 2018-09-10:
I don't see any errors, maybe try updating your graphics drivers. Does the UGS Platform visualizer work?

Baybeyya, 2018-09-10:
UGS Platform could not run because Cannot Find Java 8.1 or higher

Baybeyya, 2018-09-10:
[image]

Baybeyya, 2018-09-11:
[image]

Baybeyya, 2018-09-11:
[image]

Baybeyya, 2018-09-11:
Unsupported command

winder, 2018-09-11:
Java 8 is required, I'm not sure how you're able to run the classic UGS without it. Make sure you have the right version and you can also try the latest UGS platform.

closed 2018-09-11 by winder