winder/Universal-G-Code-Sender #1121: UGS 2.0 Starts and i get massige but i cant clik on the OK button

andreb70, 2018-09-10:
Hi,

I did download ugsplatform 2.0 and after starting the program i get a screen!

This version of Universal Gcode Sender is a nightly built.
It contains all of the latest features and inprovements,
but may also have bugs that still need to be fixed.

If you encounter any problems, please report them on github.

Now, i clik on OK noting is clik on RED X.... nothing!!

So just close and thats that. But what is the problem and how to fix this??

Thanks, Andre

winder, 2018-09-10:
Could you share the log file? You can find it's location here:
http://winder.github.io/ugs_website/guide/troubleshooting/#property-files

andreb70, 2018-09-10:
Sorry dont have a log there is non.
I have it on 2 computers Windows 10 32bit and Windows 7 64bit
And this software isnt installed i unziped it and started ugsplatform.exe on one and ugsplatform64.exe on they other both the same problem.

andreb70, 2018-09-10:
I found a log but not were you did say it was at C:\Users\I5\AppData\Roaming\.ugsplatform\dev\var\log

I hope this gives you a idea what wrong?

-------------------------------------------------------------------------------
>Log Session: Monday, September 10, 2018 4:22:42 PM CEST
>System Info: 
  Product Version         = NetBeans Platform 8.1 (Build 201510222201)
  Operating System        = Windows 7 version 6.1 running on amd64
  Java; VM; Vendor        = 1.8.0_20; Java HotSpot(TM) 64-Bit Server VM 25.20-b23; Oracle Corporation
  Runtime                 = Java(TM) SE Runtime Environment 1.8.0_20-b26
  Java Home               = C:\Program Files\Java\jdk1.8.0_20\jre
  System Locale; Encoding = nl_NL (serverlaunch); Cp1252
  Home Directory          = C:\Users\I5
  Current Directory       = D:\cnc\ugsplatform-2.0.0-stable\ugsplatform
  User Directory          = C:\Users\I5\AppData\Roaming\.ugsplatform\dev
  Cache Directory         = C:\Users\I5\AppData\Roaming\.ugsplatform\dev\var\cache
  Installation            = D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform
                            D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\ide
                            D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\extra
                            D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\java
                            D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\ugsplatform
                            D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform
  Boot & Ext. Classpath   = C:\Program Files\Java\jdk1.8.0_20\jre\lib\resources.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\rt.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\sunrsasign.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\jsse.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\jce.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\charsets.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\jfr.jar;C:\Program Files\Java\jdk1.8.0_20\jre\classes;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\access-bridge-64.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\cldrdata.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\dnsns.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\jaccess.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\jfxrt.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\localedata.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\nashorn.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\sunec.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\sunjce_provider.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\sunmscapi.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\sunpkcs11.jar;C:\Program Files\Java\jdk1.8.0_20\jre\lib\ext\zipfs.jar
  Application Classpath   = D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\boot.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\org-openide-modules.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\org-openide-util-lookup.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\org-openide-util-ui.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\org-openide-util.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\boot_ja.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\boot_pt_BR.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\boot_ru.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\boot_zh_CN.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-modules_ja.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-modules_pt_BR.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-modules_ru.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-modules_zh_CN.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-lookup_ja.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-lookup_pt_BR.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-lookup_ru.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-lookup_zh_CN.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-ui_ja.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-ui_pt_BR.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-ui_ru.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util-ui_zh_CN.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util_ja.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util_pt_BR.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util_ru.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\lib\locale\org-openide-util_zh_CN.jar;C:\Program Files\Java\jdk1.8.0_20\lib\dt.jar;C:\Program Files\Java\jdk1.8.0_20\lib\tools.jar
  Startup Classpath       = D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\core\asm-all-5.0.1.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\core\core-base.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\core\core.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\core\org-netbeans-libs-asm.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\core\org-openide-filesystems-compat8.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\core\org-openide-filesystems.jar;D:\cnc\ugsplatform-2.0.0-stable\ugsplatform\platform\c [... cut ...]
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding control state listener.
INFO [null]: Last record repeated again.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding control state listener.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding control state listener.
INFO [null]: Last record repeated 2 more times.

winder, 2018-09-11:
This is probably the issue, but I'm not sure exactly what it means:
```
java.lang.IllegalStateException: no splash screen available
```

2 things to try:
1. Re-download UGS. Occasionally a corrupt download will cause strange issues.
2. Delete the properties directory which you found here: `C:\Users\I5\AppData\Roaming.ugsplatform`

andreb70, 2018-09-11:
Oke, i wil try.

Option 2. did not work on the windows 7 Pc i have the same problem but there is a screen at the back that says.

Universal Gcode Sender is not fully localized for this language.
Please visit
http://winder.github.io/ugs_website/contrib/translation/ 
to see how you can help.

But if i go there i get a 404 File not found

Going to test option 1. now.

andreb70, 2018-09-11:
Option 2. didnt work same problem.

winder, 2018-09-11:
The current file built at 10:27: https://winder.ci.cloudbees.com/job/UGS/

Has the following MD5:
```
$ md5 ugsplatform.zip
MD5 (ugsplatform.zip) = 5f3faa74557ad474dbc53dc3115138a0
```

winder, 2018-09-11:
It seems pretty unlikely that it is a corrupt download though. Are you using a non-english locale? Maybe try launching with the classic GUI and setting the language back to english, then try the platform again.

andreb70, 2018-09-11:
I downloaded the file from https://winder.ci.cloudbees.com/job/UGS/
This gives me the same problem. Start as admin is the same. Changed locale to 4 difrent countrys stays the same.  It looks like there is some problem in the program but if thats the case others my have that same problem but thats not so.

breiler, 2019-12-12:
Is this still a problem?

andreb70, 2019-12-12:
No its no problem anymore... thanks

From: Joacim Breiler
Sent: Thursday, December 12, 2019 8:07 PM
To: winder/Universal-G-Code-Sender
Cc: Andre B ; Author
Subject: Re: [winder/Universal-G-Code-Sender] UGS 2.0 Starts and i get massige but i cant clik on the OK button (#1121)

Is this still a problem?

closed 2019-12-12 by breiler