winder/Universal-G-Code-Sender #1122: Arduino not responding to UGCS commands

jiraja03, 2018-09-11:
Hello everybody,

I would like to kindly ask you to help me please.
I have got a problem with communication between my computer (64bit Windows 7 using UGCS 1.09) and my diy CNC mill (installed GRBL 1.1f on ArduinoUNO clone with also installed CH340 USB driver).
When I connect it to my PC, it recognizes it as a usb device (Windows says it works fine) and I can open it in UGCS as well (baud 115200 8-N-1). However I can not start communication with the arduino via UGCS - firstly, it does **not show active state** and secondly, when **I send command** (for example $ or $$) **it does actually nothing** (no response nor any warning).
Where could be the problem, please??? I would be grateful for any suggestions.

[image]

Some additional info:

Sketch was succesfully uploaded to my arduino (although there was warning, that in directory /nuts_bolts.h were some redundant definitions 
#define false 0
#define true 1
#define max(a,b) (((a) > (b)) ? (a) : (b))
#define min(a,b) (((a) < (b)) ? (a) : (b)) etc.
I have no experience with programming, but here (https://forum.arduino.cc/index.php?topic=427789.0) it was discussed and may not be real problem (I DONT KNOW FOR SURE)
and there was a message, that there may be **problem with stability** due to almost full memory of arduino (30562 bytes of total 32256 bytes)

I double checked for possible HW problems (shortcuts, correct position of the shield and stepper drivers, power supply voltage) and actually found nothing - HW is likely not the problem as the PC recognizes usb device, it may be at least partly working.

THANK YOU

carneeki, 2018-09-11:
Are you able to connect to GRBL using a terminal emulator like PuTTY or
HyperTerminal, and connect using the same settings? (or if you have a linux
machine, try any of screen/cu/tip/miniterm.py to do the same)
The CH340 chips have been known to cause some problems sometimes, and I
think it's been documented in the GRBL readme or the wiki at GitHub which
is why the non-clones get recommended for use.

As for compiling GRBL, I've not successfully used the Arduino IDE to
compile it (only linux command line, and even then, I just rely on a
slightly modified Makefile).

I also note that you're using a 2015 build of UGS, it's entirely possible
that the serial protocol changed at some point between 0.9 and 1.1f. So,
the first thing I'd try is connecting to GRBL with a terminal emulator, and
if it works, I'd upgrade UGS to a more recent version and see if that fixes
the problem. Upgrading might mean you need to clear the UGS preferences if
you suffer strange UI behaviour.

jiraja03, 2018-09-11:
Many thanks for your answer.

I don´t have Linux - I am unfortunately  using Windows... Anyway, I am able to connect it to PuTTY at 115200 and it responds with the version of GRBL namely 1.1f
[image]

As you recommended, I downloaded (both) 2.0 versions and gave them a try, but unfortunately the problem remains the same in both  UGS 2.0 Classic and Platform version. No response to commands. (The 2.0 version does not even display in the console window, that the command was sent.) 
However  I ´ve noticed, that UGC Platform features Window > UGS diagnostics. I tried it and the output was like this:
[image]

Would it be possible for you to find out where the problem is from this UGS diagnostics output?
If not, would you suggest me buying the original version of ArduinoUno?

winder, 2018-09-11:
Thanks for providing the diagnostics. In this case, it seems that GRBL is unresponsive, so the UGS states aren't going to help.

Like @carneeki mentioned there have been many cases reported where arduino clones do not work with GRBL properly, specifically the CH340 chip. Although it is possible that you'll get lucky with a clone, getting a name brand arduino will save you some headaches.

jiraja03, 2018-09-11:
Dear @carneeki and @ winder,
I am really grateful for your pieces of advice. I think I am quite good at diy electronics, but definitely not at programming and SW things. Solving or even identifying problem on Windows or Arduino are quite demanding for me at the moment. I don´t think I would be able to move with it somehow by myself,, so one more thank you both guys. I would try buying the new ArduinoUNO and see what happens. Hopefully it will finally work.
Thanks and good bye

AbySet, 2018-09-19:
How to flash into Arduino Uno ?
===============================
- Put grbl folder into "Arduino\libraries" folder
- Load Arduino sketch from "Arduino\libraries\grbl\exemples\grblUpload\grblUpload.ino"
- Flash it into Arduino

and:
**problem with stability due to almost full memory of arduino (30562 bytes of total 32256 bytes)**
when you get this error when flashing, it's actualy not an issue, it happend to all arduino uno users and i'm using it and my machine work just fine.

Please let us know if you buy an original arduino uno, i can help with hardware and programing.

breiler, 2019-03-02:
If you need any more assistance just reopen this issue.

closed 2019-03-02 by breiler