winder/Universal-G-Code-Sender #1124: Importing project to netbeans

jessiwyne, 2018-09-15:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

Hi, I am new in the java world and trying to learn to developsomething. I am facing a little problem in loading this project in netbeans. I have just downloaded this IDE and never used it before, can someone help me.  I have done few steps though opened the src file in the net beans ide, and attempted to compile them, it gives me an error of Failed to execute goal org.codehaus.mojo:nbm-maven-plugin:4.1:cluster (default-cli) on project ugs-platform-branding:

winder, 2018-09-17:
You'll need to extract the zip first. There is some documentation here: http://winder.github.io/ugs_website/dev/getting_started/

jessiwyne, 2018-09-18:
Thank you winder can now access and compile the classic gui but still stuck on the ugs platform. Before starting up are we allowed to rebrand the interface. I would like to add some interesting features too.

breiler, 2019-03-02:
How are you stuck with UGS platform? What were you trying to do?

closed 2019-06-12 by breiler