winder/Universal-G-Code-Sender #1125: Stream to remote TCP ports as if it were a serial port.

carneeki, 2018-09-18:
## Feature request
Ability to stream to remote TCP ports as if it were a serial port.

Rather than typing in "/dev/ttyS0" for the port, one could type a hostname or IP address, and instead of a baudrate, one could type the remote port. A proposed NetworkCommunicator (perhaps extending a BufferedCommunicator) would then connect to the remote port and handle comms in place of a serial port.

### Possible use cases:
* Stream out to 3d printers (or other potential devices) that are network connected.
* I personally will be running a daemon on my Raspberry Pi 3 to share the serial port over TCP (inside of an SSH tunnel for security, but this isn't a part of the feature request), and this will talk to my Grbl board (UGS on Pi without the Visualiser just doesn't appeal to me, and the Pi just feels sluggish).
* There are a number of serial to network adapters which can be employed.
* Stream to multiple machines over network with multiple instances of UGS running.

### Potential problems:
None that I see. Select a mechanism for doing any serial to network conversion should be left to the user to select a server/tool and to have that establish and maintain a serial connection. I am not suggesting a server be built to do this as there are several great choices available. I propose UGS also become a "telnet client" the same way it is already a "serial client".

carneeki, 2018-09-18:
Thanks @winder ! I'll see what I can do about making a PR if I get something working.

carneeki, 2018-09-18:
I have got something that compiles, and I added an entry in the UGS options panel to use my new TCP connection driver, but for some reason UGS keeps reverting back to the last used driver.

I've pushed my changes to: https://github.com/carneeki/Universal-G-Code-Sender/tree/tcpdriver 
As it's getting late, I will take a look some more tomorrow.

carneeki, 2018-09-19:
Setting driver to "TCP" in the JSON settings file seems to get the driver to fire up, I can keep working on it :)

carneeki, 2018-09-19:
Hi @winder I've got some communication between a remote host and UGS!

I just moved steppers on my Grbl board through a Pi3, and it was updating the MPos whilst doing so :D 

As mentioned above, I had to specify the connection driver in the .json file - if you have any ideas about what I'm doing wrong with the settings, a PR might be near.

thawkins, 2018-11-17:
chillipper has a JSON server that allows remoting of the grbl conection to another machine. this allows a small SoC like a headless raspberry pi mounted inside the CNC rig to effedtivly convert the cnc into a network connected device, either via ethernet or wifi.

see: https://github.com/chilipeppr/serial-port-json-server

carneeki, 2018-11-17:
Hi Tim,

It sounds like what you describe is what I implemented :) It's probably worth linking the pull request, it's PR #1127 . I used `socat`, which I think comes with Raspbian by default, and it's a lightweight general purpose IO redirector; if it's not bundled with the OS, I probably installed it by `apt-get`.

@winder have you had a chance to review the codebeat issues? I think they're the only things that would prevent this PR from being merged.

winder, 2018-11-25:
@carneeki I didn't realize you were waiting on me! Just clicked the merge button.

closed 2018-11-25 by winder