winder/Universal-G-Code-Sender #1126: Jog -X-Y runs away

tjburdsall, 2018-09-18:
Using the jog controller, when you hold the -X-Y control for several seconds -X-Y direction locks on as it should.  In major X,Y,Z directions clicking the field stops motion.  However -X-Y motion can not be stopped, the machine runs away.
### Expected Behavior
Clicking the -X-Y feild should stop motion.
### Actual Behavior
<the machine runs away

### Steps to Reproduce the Problem
 1.  Open the program
2. connect to machine (milling machine in my case)
3.  Open Jog Controller
4.  click and hold -X-Y for about 3 seconds
5.  -X-Y travel should begin and continue after releasing mouse click.
6.  Attempt to stop motion by clicking -X-Y field  (fail)

### Specifications
#### Version
Universal G-Code Platform Nightly 9/9/18. Running on Windows 10 ver 10 running x86 CP1252 en US

AbySet, 2018-09-19:
Try it with version sep 16 2018, i tested it and it's not doing that to me.

breiler, 2018-09-22:
@tjburdsall could you please post the output from the configuration command `$$`.

I noticed a similar issue while implementing the continuous jogging. In the GRBL documentation it says that jog commands should be done using frequent small steps. I believe that when holding the jog button it will keep queuing jog commands and will eventually fill the command buffer (I'm not sure if it's in the ´BufferedCommunicator.java´ or internally in GRBL). When releasing the button a cancel command should be sent immediately clearing all buffers, but because of the filled buffer(s) it's not able to process that command quickly enough. So I made the step size pretty big and it sends the continuous jogging commands pretty seldom.

So I'm suspecting that if your settings for `$100, $101, $102` are low enough, it would queue enough commands, not being enable to process the cancel command in time.

A possible fix for this is checking the size of the command queue and make sure it's not filled before adding a new jog command (this was not possible when I implemented the feature but should be possible now with the latest additions to the code).

terjeio, 2018-09-23:
Perhaps [this](https://github.com/terjeio/grbl-Mega) patched grbl-Mega solves the problem? The input buffers (both serial and line) are flushed on receiving the jog cancel command...

tjburdsall, 2018-09-25:
Sorry for delay in response.

$100-102 values are all 200,000.

  I am not a programmer and have no knowledge of how this program works.  I
do have a question.  The system runs away when -X-Y is held but not when
individual axis X or Y is held.  Is this a clue to findind the cause?

Tom Burdsall

PR #1293 (2019-10-26): Fixed problem with continuous jogging hanging
This is a potential fix for the problems with continuous jogging hanging. I've moved the code for continuous jogging to its own handler which is responsible for checking if the command has entered GRBLs planner before issuing a cancel jog. It's not pretty but it looks like it's working.

Fixes #1126 and #1193

Thanks to @AlexHolden, @bdurbrow and @zeevy for helping out!
closed 2019-10-26 by breiler