winder/Universal-G-Code-Sender #1128: Need rise up z axis before return to zero

AbySet, 2018-09-19:
## Feature request
Can you please add something so the machine will always rise the bit (z axis) before moving to zero location when we click on [Return to zero] ?
Sometimes i cancel the task and at that time z axis still down, it can happend so i click to return to zero when i forget to rise the z axis, just for a machine safety, we may need the machine automaticly rise z before go to zero.
Thank you so much.

### Specifications
#### Version 2.0 [Nightly] / Sep 16, 2018
 
#### Operating system
Windows 8.1 64 bit

#### Platform
GRBL 1.1

carneeki, 2018-09-19:
I had a similar issue, to resolve it I created a macro that:
G53 G0 Z0; G90 X0 Y0 F600

(G53 temporarily tells the G0 command to use the machine coordinate system;
depending on how your Z axis is set, this may be 0 or it may be a high
number like 150mm)

AbySet, 2018-09-19:
Thank you for reply, if i'm not mistaking, you should click on macro button to run the code, if this is the case, i want it to do that automaticly cz like i said, it happend to me to forget.

breiler, 2019-02-19:
@carneeki your example requires that you have homing enabled, right?

The "return to zero" button should return to the work position 0 with the current implementation:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L196-L206
- Move to Z0 (top of the material) if it's lower than 0, 
- Move to X0 Y0
- Move to Z0 (only applicable if we where higher than 0)

I had issues with this when my material wasn't completely leveled and the bit scratched the surface. 

I've previously suggested that we should add an option to move to a configurable Z safe height. Some have also suggested that the "return to zero"-command should be completely configurable. 
We now also have the possibility to check if homing is enabled and then move to the machine zero.

carneeki, 2019-02-20:
@breiler - I'm not sure, I wouldn't want to try it without homing and
limits just in case one axis travels too far.

Safe Z heights would be great!

AbySet, 2019-02-20:
Thanks for the answer breiler. This explain everything.

PR #1357 (2020-02-27): Added setting for safety height
Added setting for safety height when returning to home. Moved the returnToHome implementation to the abstract class as the implementation will work on all controllers. Added the "return to home" as a capability instead of letting the controller throw an exception if not implemented.

Fixes #1128
closed 2020-02-27 by winder