winder/Universal-G-Code-Sender #1129: Various Issues

Preston4, 2018-09-20:
Hello everyone, I am working on getting my schools cnc router set up, however, I am facing obstacle after obstacle. Ok so I am able to manually move the machine + ( I’m using universal g code sender) on the z axis, and nothing else on any other axis( the other axis never move. when I try negative, the machine makes different really high pitched noises, they change depending on how many times I press the command. I am working on the HX8410. Also, the stepper motors get REALLY hot, even though they are not moving.

here is the stuff the machine prints
>>> $H
An error was detected while sending '$H': ALARM: Homing fail. Streaming has been paused.
Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 7    (dir port invert mask:00000111)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 1    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 1    (hard limits, bool)
$22 = 1    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 1500.000    (homing feed, mm/min)
$25 = 1500.000    (homing seek, mm/min)
$26 = 50    (homing debounce, msec)
$27 = 5.000    (homing pull-off, mm)
$100 = 787.400    (x, step/mm)
$101 = 787.400    (y, step/mm)
$102 = 787.400    (z, step/mm)
$110 = 1000.000    (x max rate, mm/min)
$111 = 1000.000    (y max rate, mm/min)
$112 = 1000.000    (z max rate, mm/min)
$120 = 50.000    (x accel, mm/sec^2)
$121 = 50.000    (y accel, mm/sec^2)
$122 = 50.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)
ok
>>> $G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0.0 S0.0]
ok

winder, 2018-09-20:
That could mean your stepper motors aren't wired incorrectly

AbySet, 2018-09-21:
- Did you mouted the machine yourself ? did you wired it ? if so then check motors wires one by one, if they are correct then check contact on wires from end to end by using multimeter (disconect them to do that).
- Did you noticed that the move direction is inverted on your settings ? see $7.
- Also the values on $100, $101 and $102 sounds wiered to me but this have nothing to do with the issue you have.

Preston4, 2018-09-21:
I did not wire the machine myself, it comes already assembled. Should I just manually reverse the worm drives to where the machine is in the center of the workspace? I think it might have crashed and can’t reverse from it. Any ideas?

winder, 2018-09-21:
It sounds like you should contact the vendor, they'll have much more knowledge about your machine.

AbySet, 2018-09-21:
- According to your settings, you are using limit switchs, $21 is enabled, i'm suprised to know that your machine has crashed.
- Again, i need you to **notice that your motors direction is inverted in settings, see $3**, to make it back to standard direction set $3 to 0 but before you do that, see if you can manualy move X and Y axis to get them away from the crash position (with power off preferably) and then power it on and set $3 to 0 and do a homing cycle. 
- Notice that when the machine limit switchs are trigered (pushed or released), grbl will report an error, that is normal, just do soft reset and unlock when that happend so you can use jog buttons again.
If your machine came with wires plugs to the controler, please check if they are pluged where they belongs to.
- If all this did not work, as Winder said, you need to contact the vendor.

(A very last think to suspect, the conrtoler motors drivers may be difective and need to be changed).

Preston4, 2018-09-21:
Thank you all some much, I compleatly took apart the machine, and found that the worm drives are almost compleatly seized. I can move it by hand, and need to use a wrench to move it. Any ideas on why this is? How do I fix?

AbySet, 2018-09-21:
Disconect motors wires, this will make it much easier to move by hand, when they are connected they resist to the move due to the electromagnetic action they generate current that will try to hold the move and this is all normal behaviour.

Preston4, 2018-09-21:
I already have done that, the belt isn’t even connected to the motors, the screw itself is seized,( like the bearings are stuck or broken.

AbySet, 2018-09-21:
Dam, this is rearly an issue, sorry for that.
Anyway i did some reaserch and i found this:
[image]

And i recommand you to use those settings instead:
[HX8410 Settings.txt](https://github.com/winder/Universal-G-Code-Sender/files/2406125/HX8410.Settings.txt)

Also i found this video that shows what settings the guy is using:
https://www.youtube.com/watch?v=muSIo5jvFB4

I think you were using double speed of what your machine can handle and that causes the machine getting stock and making strange noise (look at $100, $101 and $102 values).

Hope this help.

Preston4, 2018-09-21:
Thank you so much, I’ll be with the machine later today probably and will check everything out, for now, thank you.

breiler, 2019-10-27:
I'm closing this as it doesn't have so much to do with UGS.

closed 2019-10-27 by breiler