winder/Universal-G-Code-Sender #1130: GRBL stops responding after first 10 commands

arjtilley, 2018-09-25:
For example, if I send the command G91 G0 X1 ten times from UGS, GRBL stops responding. Can be any command. Doesn't matter if typed in manually or with a macro or jogging buttons. Then I have to do a soft reset to regain control.

If I send the same commands G91 G0 X1 manually, not using UGS, e.g. with Arduino IDE serial monitor, I do not have this problem.

breiler, 2018-09-26:
What version of GRBL and UGS are you using? What operating system are you using. Can you also post your GRBL configuration.

Is the Arduino a genuine or a cheaper copy?

arjtilley, 2018-09-26:
GRBL v0.8
UGS Version 1.0.9
Windows 7 Ultimate 64-bit

Its some generic Arduino Nano clone, based on MEGA328P IC. Don't know if it has Arduino boot loader or not.

I am using the default config that was already on the board. I am very new to GRBL, so there may be an incorrect setting that I have not noticed.

`$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=1 (status report mask:00000001)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=1200.000 (x max rate, mm/min)
$111=1200.000 (y max rate, mm/min)
$112=600.000 (z max rate, mm/min)
$120=50.000 (x accel, mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=400.000 (x max travel, mm)
$131=400.000 (y max travel, mm)
$132=400.000 (z max travel, mm)
{0/0}ok
`

arjtilley, 2018-09-26:
Here is the grbl config:
[grblconfig.txt](https://github.com/winder/Universal-G-Code-Sender/files/2421365/grblconfig.txt)

AbySet, 2018-09-29:
First get the latest version of ugsplatform and see if the issue get resolved and update java to 1.8 version.
You may also get grbl 1.1 version and flash it into arduino. (not quiet sure if nano version of arduino have something to do with the issue).
If you want to make a delay between the commands then use G04 P0.1 for 0.1 second delay.
You may consider checking if your serial port baudrate is set to 115200 after you get the latest version of ugsplatform (older versions runs with 9600 baudrate).
Notice that your $1 is set to 255, that keep your motors powered all the time and get hot even when they are used.

breiler, 2019-02-20:
Closing this issue as there was no response from the issuer.

closed 2019-02-20 by breiler