winder/Universal-G-Code-Sender #1131: error while processing response

ZHIQIU-ANSON-CHEN, 2018-09-28:
Have no idea why this issue occur, it was working well in the morning.
I'm using arduino and grbl as arduino lib. Arduino connection are fine.
After I click connecting pattern to connect. This occurs.
Could someone help me out?
Appreciated!
[image]

AbySet, 2018-09-28:
Do you have original arduino ? some ppl are reporting bugs with none original ones.
Also try to reflash grb into the arduino.
But before you go for reflashing check your baudrate:
 In windows serial port better (but not must be) at 115200.
 In ugsplatform it must be at 115200.

ZHIQIU-ANSON-CHEN, 2018-09-28:
> Do you have original arduino ? some ppl are reporting bugs with none original ones.
> Also try to reflash grbl into the arduino.
> But before you go for reflashing check your baudrate:
> In windows serial port better (but not must be) at 115200.
> In ugsplatform it must be at 115200.

Yes, I purchased it from Electronics store. It is guaranteed. I did reflash grbl into the arduino and serial is 115200 as well. It was working fine in the morning.
Have no idea somehow it just turn to this issue.

AbySet, 2018-09-28:
I'm not mac user but i did some reaserch for you and i found that someone with the same issue got it back to work in mac (using name Mitch-X) and he said this:

`For those of you that are interested, I went through and systematically replaced all component and updated and reinstalled all the relevant software. Only upon removing the installed USB-Serial Drivers and replacing them with the updated CH34x Drivers for Mac, did the issue cease.`

the link to the thread is: https://github.com/winder/Universal-G-Code-Sender/issues/974

hervema, 2019-01-02:
Hello. I've exactly the same issue, whatever version I'm using, last stable or last nightly, and whatever UGS Classic or UGS Platform ! :-( I'm using an official Arduino, installed last version of Java and using GRBL 0.9j on a Windows 10 last release. Note that version I'm using until now was UGS 1.09 and it works perfectly with the same CNC...   Let me know if you need additional information to help you to resolve. Thanks in advance. Regards.