winder/Universal-G-Code-Sender #1132: Right Click on Jog causes infinite loop

AlAmantea, 2018-09-29:
Once in UGS Platform and connected, go to jog panel.

Right click on jog buttons and machine (X Carve GRBL 1.1f) will go into an infinite jog loop until you disconnect from the machine!

AbySet, 2018-10-01:
I tested it and i confirm that.
But:
It's acruatly a good feature that we should keep cause if you left click on same direction it will stop the loop.
I also was playing with different combinations and i found that if you do'nt left click at the same direction after the right click sometimes the loop wont stop.

My Specs:
- Windows 8.1 x64
- grbl 1.1
- ugsplatform nighly build 09-23-2018.
- java 1.8
- Arduino uno shield controler.

AlAmantea, 2018-10-01:
I don't think I would keep it. Too much chance of a machine crash with it the way it is. It makes more to allow a repeat on left click and hold until released.

AbySet, 2018-10-01:
Yeah, holding left click is doing the same, you'r right, then i think they should remove right click action.

carneeki, 2018-11-30:
This weekend I'm going to look into both this and #1146 as I suspect they are related and a bit of a worry for me too.

Just so I'm clear on the understanding of desired usability:
* While holding left mouse button (LMB), machine will jog
* When we let go of LMB, machine will stop jogging
* If RMB (right mouse button) is pressed: ignore and do nothing.

AlAmantea, 2018-11-30:
Yes, that would be the desired operation.

Minimum would be to disable right click on jog buttons so it doesn't do anything on RMB.

carneeki, 2018-11-30:
@winder This sounds like good idea but a bit of a bigger change than I was hoping to make here [I'm concerned I'll break `getCapabilities()` in other places], especially since I'm still pretty green to the codebase.

At present, I'd like to focus on just the LMB and RMB events and desired actions, and break the world next weekend ;)

commit 1564b908 (2018-12-01): ignore RMB with regards to LongMouseListener, fixes #1132, fixes #1146
commit e13290c3 (2018-12-01): Merge pull request #1157 from carneeki/master

ignore RMB with regards to LongMouseListener, fixes #1132, fixes #1146
closed 2018-12-01 by winder