winder/Universal-G-Code-Sender #1134: Probe

waltoncnc, 2018-10-02:
--

## Problem description 
How do I go back to a previous version of nightly builds? The new version we are using the probe feature is somewhat working. When we uninstalled the new version and installed older version it is like he newer version is still there. The probe is not working right.

### Specifications
#### Version
<!
  
  UGS Platform 2.0 - Nightly Build May 18,22018
-->

#### Operating system

#### Platform

AbySet, 2018-10-02:
I do'nt have any problem with prob but seems like you are already using an old version, the newest one has been released on 09/23/2018.
Here are steps to follow for clean new install assuming that you are using windows 8.1 (you did not mention whitch system you are running):

A . Deleting deleting usgplatform cash (including settings):
- Goto folders setting and make hidden files visible.
- Open [Users] folder.
- Open [Your user name] folder.
- Open [AppData] folder.
- Open [Roaming] folder.
- Delete [.ugsplatform] folder.

B . Replacing ugsplatform version:
- Delete ugsplatform folder you want to replace.
- Download the one you want to install
     . Last Nighly build here: https://github.com/winder/Universal-G-Code-Sender
     . Last stable release here :http://winder.github.io/ugs_website/download/

waltoncnc, 2018-10-02:
Are you using just z probe or xyz probe?

AbySet, 2018-10-02:
i'm using just z prob

breiler, 2023-08-17:
I am closing this as a lot of fixes has been made to the probing functionality. Please open a new issue if you still find problems with it.

closed 2023-08-17 by breiler