winder/Universal-G-Code-Sender #1136: UGCS crashes when an incomplete program is loaded

infernozx, 2018-10-03:
## Feature request

## Problem description 

UGCS crashes when a file is loaded that does not have a basic movement. 
### Expected Behavior

Some kind of warning, or note that informs the user that the program is invalid.
### Actual Behavior

Program locks up, needs to be killed via the task manager
### Steps to Reproduce the Problem

  1. Create a file. Example:
F010 F400 G38.2 Z-10
F020 F400 G38.4 Z50
F030 G92 Z0
  2. Load this program into UGS, it will crash.
  3. Add a line at the end of the above program:
F040 G00 Z12
  4. Run this program. It should load successfully.

### Specifications
#### Version

UGS Platform 2.0 Stable
#### Operating system

Lubuntu

#### Platform

GRBL 1.1
This functionality can be tested without any controller connected.

#### Other

I'm not sure what will stop it from crashing besides an actual movement command. I understand that G38.2 is a movement command (probe) but from my research it seems that I need to have a standard X, Y, Z movement command in the file to prevent the crashing.

AbySet, 2018-10-04:
I tested your example and it's not crashing for me, i got this error message when trying to open the file with ugsplatform:
_Problem opening file : Multiple F-codes on one line._

My specs:
Windows 8.1 x64
grbl 1.1
ugsplatform nighly build (released 09-23-2018)
java up to date (i thinik 1.8)

infernozx, 2018-10-09:
Could be platform specific? Although from my rudimentary understanding it shouldn't be...

AbySet, 2018-10-09:
Not quiet sure about that i'm not using linux but hmm yeah it could be. We should ask people who are using linux or owner.

nkuck, 2018-10-13:
I’m new to ugs and just trying to get some g-code loaded but keep getting the “multiple f-cods” error message. Running on a Mac with no steppers installed yet. I get the error when I just try to connect to my board or when trying to open a g-code file. Where should I focus? Thanks for some noob guidance.

AbySet, 2018-10-13:
nkuck, you are supose to post your own issue but anyway i'll try to help.

As far as i know, a "multiple f-codes" error explain itself, that mean in one line of gcode file you got more than one "F" code and this is not allowed.
F stands for feedrate whitch we can consider a move speed, let say for instance you want the feedrate to be 500, you put F500 in the move line with the move destination cordinates, the ugs do not allow you to put two speeds in same line of gcode.
Here is an authentic line from true gcode file:
G1 X84.9705 Y79.9854 Z0.25 F3000.0
This move go to 84.9705 for X axis, 79.9854 for Y axis, 0.25 for Z axis with the speed of 3000 (feedrate).
If the line was somehing like: "G1 X84.9705 Y79.9854 Z0.25 F3000.0 F2000.0" you will get that error.

nkuck, 2018-10-13:
Here is what I was trying to load that generated the error (Sorry for not creating a new posting):
G17 G20 G90 G94 G54
G0 Z0.25
X-0.5 Y0.
Z0.1
G01 Z0. F5.
G02 X0. Y0.5 I0.5 J0. F2.5
X0.5 Y0. I0. J-0.5
X0. Y-0.5 I-0.5 J0.
X-0.5 Y0. I0. J0.5
G01 Z0.1 F5.
G00 X0. Y0. Z0.25
Can you point out the error? It was an Internet file that looked simple and harmless to try.

AbySet, 2018-10-13:
Dont be sorry about the post i'm not admin or some, i just want you to know that every type of issue is supose to be in it's post.
Ok now about your code i think there is no error on it, i have put it into a file:
 - The file load was a succes.
 - Running the code in my machine was a succes.
No errors to me.
I'm using windows 8.1 x64, ugsplatform nighly build (october 7 2018) and java 1.8.
At this point i can'nt tell what is the problem, i can just hope you'll find out with someone else. so sorry.

nkuck, 2018-10-13:
Well, thanks for looking at it. I can at least look beyond the code, so that is a start.

breiler, 2020-03-09:
Can't reproduce the error, could have been an issue with an earlier UGS build.

closed 2020-03-09 by breiler