winder/Universal-G-Code-Sender #1137: Adding support for a new board

PeterCassar99, 2018-10-08:
Hi there, I'm looking at UGS as a front end for a new advanced back end board I'm working on. I'd like to do the coding myself although the UGS Platform seems to have some peculiar habits. I'm wondering if anyone can help me understand where I'm going wrong?

The problem is that I've created a new firmware .json file in ./firmware_configs for my board and at runtime it fails to locate the file. When debugging the code the enumeration for the supported boards doesn't load my board. I can see the new config copied to .ugs/firmare_configs on the local file system.  

The UGS core version seems to locate the firmware and connect to my board.

What is the best way to start setting up the support for a new board?

winder, 2018-10-08:
Is there an exception being thrown when you launch UGS Platform?

In addition to the json file you'll need to create an enum in `ControllerSettings.java` with your new type and update `getController()` to instantiate your backend controller object.

UGS Classic and UGS Platform both use the same code for this. Sine the classic version is working for you, it's possible you have some stale files in the Platform version and a clean build will fix the issue.

PeterCassar99, 2018-10-08:
Thanks Will, Your suggestions helped. I must've had some very stale files in the UGS_platform module . Now the system picks up the firmware file and connects to the board . I was previously building the modules individually from netbeans and building from the UGS_Parent level appears to have cleaned everything out and set things up correctly. 

Off to have some fun now  to implement the special features.

closed 2018-10-08 by PeterCassar99