winder/Universal-G-Code-Sender #1140: Will not run on macOS 10.12.2

clarkdv, 2018-10-12:
macOS 10.12.2 and Java 8 Update 181 installed.

When I launch the Stable version of UGS Platform I get the error below in Terminal.
I have been able to get the nightly build of UGS Platform running but I need to be able to run the Stable version.

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/Users/cnc/Desktop/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

clarkdv, 2018-10-19:
Mine doesn't open, it just quits after I get the error message.

breiler, 2019-02-20:
It looks like it's running Java 9 or later. UGS requires Java 8, make sure you have the right version activated.

See this for instructions:
https://stackoverflow.com/questions/21964709/how-to-set-or-change-the-default-java-jdk-version-on-os-x

I'm closing this issue as it's quite old, feel free to open again if the issue persists.

closed 2019-02-20 by breiler