winder/Universal-G-Code-Sender #1142: TinyG jog not working

CodeZone8, 2018-10-25:
The jog panel is not enabled after connection to the controller. 
Everything else works fine.

 UGS Platform 2.0 - Nightly Build Mar 23, 2018
 Windows 10
 TinyG

No way no jog machine other than a right click on the visualizer and Jog To x,y

winder, 2018-10-25:
Try the latest nightly build, there have been a lot of TinyG improvements this year.

CodeZone8, 2018-11-25:
> Try the latest nightly build, there have been a lot of TinyG improvements this year.

I did used the latest build at that time. Still no go. A quick check shows there is nothing new since my last post so I guess I am out of luck using this software until there is a fix for this.

norru, 2019-09-01:
Why was this bug closed?

norru, 2019-09-01:
Opened https://github.com/winder/Universal-G-Code-Sender/issues/1265

PR #1300 (2019-11-02): Fixed tinyg support
TinyG is now usable, however some issues are still left: 
Fixes #1142 and addresses some of the issues in #1262
closed 2019-11-02 by breiler