winder/Universal-G-Code-Sender #1143: Freezing up

waltoncnc, 2018-10-25:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

AbySet, 2018-10-26:
Can you please put some text about your problem se we try to help ? thanks.

waltoncnc, 2018-10-26:
Freezes up during carving. Have to shut everything down and start over. Have cancelled,router and computer on their own dedicated breaker. Have tried it on 3 computers. Get it cutting again and within a few seconds it does it again.

AbySet, 2018-10-26:
Is it software freeze or machine freeze ?
We also would like to know:
- Your operating system.
- Your software version.
- Your grbl settings.
- Your machine controler (In the electronics box)
- Your steeper motors (nema17, nema23 or so).
- If possible the last grbl command before the freeze on the gcode file.

waltoncnc, 2018-10-28:
Software freezes up.
Windows 10 and Windows 7
Grbl 1.1f
It is a stock Shapeoko 3xxl.
Tried three computers and everything is on its own dedicated breaker.

AbySet, 2018-10-28:
If you tried with the same usb cable into all 3 PCs then try with some other usb cable.
If you tried with different usb cables and still doing the same issue, then you may call service for controler replacement, hope you still have the waranty for that.

This assume that:
- All wiring are correct.
- Arduino driver properly installed.
- No virus on your PCs.

Please let me know when you fix it and how.
Good luck.

BarrettLowe, 2019-04-22:
I am using the nightly build on a shapeoko 3xxl using Windows 7 home SP1 and the nightly build from December 18. All seems fine to me. I've got merely 4 gigs of RAM. Maybe some of this can help you?

AbySet, 2019-04-22:
If it does freeze always with the same file and only with that file and according to your previour posts i suspect:
- Unsuported gcode line by grbl (then you should have an error message on screen).
- Overspeeding code on a certain line of gcode file (F code) that pushs the speed of steper motor more than it can handle.

Can you post you gcode files ?
Does different gcode file the same issue ?

(All this assuming that the machine is freezing not the software).

carneeki, 2019-04-25:
A HOWTO for @AbySet 's suggestion that step frequency could be too fast, let's confirm or rule it out by getting the following information:

1. The last 5 or 10 lines of code that cause the failure (or possibly further up if the F word is set earlier).
2. Output of the Grbl config, this can be copy+pasted from typing '$$' into the command box and pasting the output back here.

**Background reading (optional) & Explanation:**
There's a value configured inside of Grbl called the "step rate". It has a maximum limit based on a bunch of complicated maths and how the Arduino inside the Shapeoko works, but essentially, this step rate cannot exceed 30k steps per second (30khz). The stepper motors take a step pulse from the drivers (which are kind of like amplifiers for the Arduino microcontroller), and move forward a small "click" or step. If you try to make the gantry move too fast, Grbl tries to step faster and faster. Eventually it may exceed this 30khz figure, and the controller resets.

Shapeoko should have configured `$110`, `$111` and `$112` appropriately so that this wouldn't happen, or maybe the CAM software you use should prevent feed rates (F word) from exceeding the maximum rate.

Further reading (though not necessary, but might be of interest) - https://github.com/grbl/grbl/issues/194 and https://github.com/gnea/grbl/issues/41.

This all assumes that the step rate is being exceeded. We cannot tell until you give us the info in (1) and (2). Thanks!

breiler, 2019-04-26:
Could you check which connection driver you are using, try switching to the other:

[image]

AbySet, 2019-04-26:
Thank you carneeki, i also suggest to follow breiler recomandation.

breiler, 2019-11-02:
Closing as there wasn't any feedback on suggested workarounds.

closed 2019-11-02 by breiler