winder/Universal-G-Code-Sender #1145: "Initiate Probe" keymap shortcut

ione2k, 2018-10-29:
Hi All,

I was wondering if there is a way to include a shortcut for the "Initiate Probe" button? I can only find the option for "Probe Module" shortcut which takes me to the window where the Initiate Probe button is. I am looking to have this shortcut included on a keypad that I use as a pendant beside the spindle to position and zero it. It is quite uncomfortable to have to come back to the computer to press the "Initiate Probe" button to zero the Z axis which also defeat the purpose of having the keypad beside the spindle in the first place.

Thank you in advance for your help.

AbySet, 2018-10-29:
Yeah i've been looking for it and there is none, you may post this in a request section to give it better chance to be taken in consideration and may be added in future releases.

PS: I'm just a user like you.

breiler, 2019-02-24:
This should be pretty easy to implement, have a look at the probe module here:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ProbeModule/src/main/java/com/willwinder/ugs/platform/probe/ProbeTopComponent.java

Use this service for creating keymap shortcuts:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugslib/src/main/java/com/willwinder/ugs/nbp/lib/services/ActionRegistrationService.java

Have a look at this for example of usage of the ActionRegistrationService:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugscore/src/main/java/com/willwinder/ugs/nbp/core/control/RunActionService.java

AndyCXL, 2021-08-05:
Thinking about the scope of this issue...
- Probing will initiate movements on the machine that shouldn't surprise the operator, especially given the ask coming from a keypad/puck beside the spindle, and therefore hands close by too. Safety above all else.
- What can/could the user do if 'up close' to the spindle and the probing can be seen to fail, or predicted to fail (clip becomes detached, puck moves etc). Is this any worse than when initiating the action from a button click on screen?
- Should there be multiple initiation shortcuts? Z, XY, XYZ
- Is there any roadmap idea or ambition to probe for A, B or C? I can't imagine so, but worth asking
- Without having looked into the code at all as yet, I presume that if the Probe Module is not open, no action should arise from a menu or shortcut click

jayseeee, 2022-11-10:
This would be a great enhancement. I plan to build a remote controller with an Arduino Micro. The Arduino Micro can  emulate a keyboard. Would be great to have a physical button for the probe too.

breiler, 2023-08-16:
I have now added separate actions for initiating probing. For now those actions will popup a confirmation dialog the first time asking if the user really want to start the probe with the current settings.

This confirmation dialog can of course be expanded to show the probe settings or a preview of the probe operation. But I decided to keep it simple to start with.

These probe actions are also mappable to a gamepad, keyboard shortcut or can be added to the toolbox or toolbar for quick access.

This is available in the latest nightly build.

jayseeee, 2023-08-17:
Thx a lot!!!

closed 2023-08-17 by breiler