winder/Universal-G-Code-Sender #1148: strange work/machine positions

anotherboringuser, 2018-11-04:
Hi there, 
My work position and machine position show strange values. If I start at X0 Y0 Z0 with 10mm entered in the XY Step Size Box and then klick on X+ (or Y+) the work/machine position shows 9.989mm . If I klick again it shows 20mm, then 29.989mm, then 40mm, then 50.011, then 60,…. 
The machine moves as it should. 
If I press reset Zero, it displays 0.011 for the work position. The same happens to the Y values if I moved the Y Axis. 
Restarting ugs doesn’t help. 
Windows 10 64bit, GRBL 1.1f, UGS Version 2 nightly Oct 28, 2018 

[image]

carneeki, 2018-11-04:
This sounds like grbl finding the nearest step to the desired target.

Stepper motors (which the machine uses to move the table /spindle / laser
head) move in discrete "steps" or "clicks" around a circle, so there's a
limit to the resolution. This can be found in the $100, $101 and $102 settings.
Think of it as being like pixels on a computer screen to show you an image,
it's approximate steps rather than a smooth curve for the image you see.
The steppers can move to each pixel but not in between. If you find the
error amount unsatisfactory there is a technique called microstepping that
will help make it smaller but can't eliminate the stepping completely.

(Danger! Here be dragons! Be warned!)

To adjust the resolution, your stepper drivers should have a manual with
some switches or jumper settings that allow you to specify how many
microsteps per revolution on the steppers, and maybe this value can be
increased. Then the $100, $101 and $102 settings may need to be tweaked to
match the new resolution. There are several caveats to doing this, one is
that GRBL can only produce so many pulses to the drivers every second
(about 30 thousand), and if the microstepping is too high, GRBL has a
software crash and can wipe the config, and weird things happen to the
reported machine position, so it needs to be balanced, so the maximum
feedrates get reduced in $110 thru $112.

If you bought your machine commercially, I would recommend contacting the
builders and ask them about microstepping, though chances are it is already
balanced well.

If you decide to change the settings, definitely use UGS to make a backup
of the config in the Firmware Settings so you can come back to a working
state if things go wrong. Save this config on a separate USB stick and post
it to Dropbox or email to yourself for safekeeping. Best of luck!

anotherboringuser, 2018-11-04:
Thank you very much for your explanation! This seems to be the reason for the behavior. This makes sense. 
I build my machine myself, tweaked the settings  and I also enabled micro stepping (half steps), but I did not think about that because I changed my machines motion system from screws to belts, and back then the steppers where set to 160 steps per mm, so UGS always displayed the work position value that I expected. 
I apologize for starting this issue even though there is no bug in UGS related to what I experienced.

closed 2018-11-04 by anotherboringuser