winder/Universal-G-Code-Sender #1149: Spindle move distance is always multiplied

polyenios, 2018-11-06:
## Problem description 
I am sending a 1mm step through the jog controller and the spindle moves by 5mm instead of 1mm. Same multiplication happens when I send a project file. Visualizer shows the correct dimensions of the project but when executed, dimensions are multiplied for all three axes.

### Specifications
#### Version
I have tried both versions and had the same issue:  
  UGS Platform 2.0 - Nightly Build Mar 23, 2018
  UGS Classic 1.0.9

#### Operating system
  Windows 10

#### Platform
Arduino
grbShield v3

AbySet, 2018-11-10:
Did you recently removed some jumpers from arduino cnc shield ?
What is your machine specs, steeper motors?
Are you using belts or linear screw and what are their specs (if belts, what is the pouly specs in terms of number of teeths).
And after you tell us your machine specs we also need to see your grbl settings for $100, $101 and $102 (Theses values must be calculated based on pouly and belts specs or on linear screw specs).

The jumpers on arduino shield below the motors drivers are what's called microsteps settings and their values must be considered to set the right values for $100, $101 and $102. So we also need to know if you are using them and at what value they are set.

jahnj0584, 2018-11-11:
Sounds like you didnt calibrate your axis yet.

AbySet, 2018-11-17:
Hey, sorry i'm too late i was busy with other stuff.
Ok let's see your data:

- I recommand $0 to be set to 10 for the hardware you are using (arduino uno and motors drivers).
- Your $1 is set to 255, that's good to hold the motors tide but will keep them under power even when the job is finished whitch will make them being hot even at idle.
- Your $2 is set to 28 and it's maximum value is 7, this is for inverting pulse signal whitch you don't neet for the hardware you are using, so just set it to 0
- You may set $10 to 3 and that will report both the machine position and work position.
- $30 is set to 1000 and that's too low to me but it does'nt matter if you are not using arduino to controle the spindle speed then it has not effect.

Let's talk about steps per mm values:
----------------------------------------
- For X and Y axis you are giving only number of teeth on the pouley and _i need to know the pitch of the belt which is the distance between two teeth on the belt_ then i can calculate the exact value of $100 and $101. I also need to know if you are using microsteps settings on the arduino cnc shield (the jumpers under the motor drivers).
- For Z axis i only _need to know if you are using microsteps settings on the arduino cnc shield_, i would say the value of $102 is too high but let's wait for the calculation.

For the accelations values:
-----------------------------
- X and Y axis $120 and $121, they look too low to me, i recommand using 6.25% of $110 and $111  values repectivly to each axis (for example $110=500 will have better accelaration $120=31.25).
- For Z axis acceleration the value seems ok.

I will wait for the requested infos to do calculations for you, but if you wana do it yourself here is the link to follow:
https://www.norwegiancreations.com/2015/07/tutorial-calibrating-stepper-motor-machines-with-belts-and-pulleys/

Hope this help you.

polyenios, 2018-11-22:
Hi! Thanks again for your time.
Pitch for the belt of X,Y axis is 2.032mm. So with a 8x micro stepping my setting of 43.74 is ok.
For Z-Axis micro stepping should set to 2. I cannot tell from the jumpers because I have a v3 grblShield.

During my tests, X,Y axis are performing flawlessly even after several hours of operation. Z axis is initally moving with no problems but at some random point it is jumping steps. I will check with your recomendations to see if the problem is resolved.

breiler, 2019-10-27:
Closing as it seems to be an hardware/configuration issue not related to UGS. There was no further responses since nov 2018.

closed 2019-10-27 by breiler