winder/Universal-G-Code-Sender #1150: Open and send gcode file fails

macdroid53, 2018-11-16:
## Problem description 
I can't get a file to load and send to grbl.

I'm using classic UGCS on Ubuntu 18.04

If I type the following G-codes they all return ok:

G90
G21
G1 F3000

If I create a file with:

G90
G21

Browse and open it the Rows in File shows as 2.

If I add to the file:
G90
G21
G1 F3000

Then reload the file Rows in file shows 2

If I change the file to:
G90
G21
G21

Rows in file shows 3

Also, If I let it sit for a bit, UGCS attempts to the send the file and Grbl goes into Hold state. After that the cancel doesn't work, and basically have to close the connection and start over...

### Specifications
#### Version
  UGS Classic 2.0 - Nightly Build 23 2017

#### Operating system
  Linux Ubuntu 18.04

#### Platform
  GRBL 1.1g

#### Other

AbySet, 2018-11-17:
You may write G01 not G1 and give at least one axis to move:
example:
G01 X1000 F3000

breiler, 2019-03-02:
I can't reproduce this anymore, please update to a later version of UGS

closed 2019-03-02 by breiler