winder/Universal-G-Code-Sender #1151: UGS Platform crashes on start on ubuntu 18.10 with java 11

giasone70, 2018-11-18:
## Feature request

## Problem description 
Starting the program it crashes after displaying the nightly build warning.
Looking at the output in the console window you get this:

> WARNING: An illegal reflective access operation has occurred
> WARNING: Illegal reflective access by org.netbeans.ProxyURLStreamHandlerFactory (file:/home/lab/Programmi/ugsplatform/platform/lib/boot.jar) to field java.net.URL.handler
> WARNING: Please consider reporting this to the maintainers of org.netbeans.ProxyURLStreamHandlerFactory
> WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
> WARNING: All illegal access operations will be denied in a future release
> Initializing...
> Loading LocalizingService...
> Loading JogService...
> Loading ActionService...
> Loading MacroService...
> Loading SendStatusLineService...
> Loading SettingsChangedNotificationService...
> Loading WindowTitleUpdaterService...
> Services loaded!
> Setting UGP version title.
> Initializing firmware... ...
> /resources/firmware_config
> /resources/firmware_config/test_2.json
> /resources/firmware_config/smoothieboard.json
> /resources/firmware_config/grbl.json
> /resources/firmware_config/test_1.json
> /resources/firmware_config/tinyg.json
> Inconsistency detected by ld.so: dl-lookup.c: 112: check_match: Assertion `version->filename == NULL || ! _dl_name_match_p (version->filename, map)' failed!
 

### Expected Behavior
Normal program start
### Actual Behavior
Crash

### Steps to Reproduce the Problem

carneeki, 2018-12-06:
I also have problems with Java 11 (but not the same one) on Arch Linux (rolling release, well maintained & updated). Sometimes I can compile UGS, sometimes I have to roll back to Java 8, and I think mostly it's to do with NetBeans 8.x as a dependency. I suspect that there's a plan to move UGS to a more recent NetBeans version in the future, but maybe Will is just short for time.

Recommendation: try rolling back to Java 8. Here's a couple of ways for different distros, including this as it may prove helpful to others for other distros too.

| Distro | Command | Notes & Link |
| ----- | ----- | ----- |
| Debian / Ubuntu | `update-alternatives` | This answer covers both `java` and `javac` for running and building https://askubuntu.com/a/388693 |
| Arch / Manjaro | `archlinux-java` | https://wiki.archlinux.org/index.php/java#Switching_between_JVM |
| Red Hat / Fedora | `alternatives` or `galternatives` | https://fedoraproject.org/wiki/Java and https://fedoraproject.org/wiki/Alternatives_system |

If switching the Java 8 doesn't fix the problem, definitely report back here. HTH!

hvfrancesco, 2019-05-11:
same problem on Ubuntu 18.04 with java 11, reverting back to Java 8 works

alebarra, 2019-12-12:
while waiting for ugs to be updated to a more recent platform, it is still possible to launch it without rolling back to version 8 of jdk.
in debian/ubuntu, you can install more openjdk versions in the system and set any one as default, by running ugs with version 8 without performing a rollback, ugs accepts the --jdkhome parameter which specifies the jdk directory with which it can be executed:

`ugsplatform --jdkhome {path to openjdk8}`,

 for example, in my debian distro the command become:
`ugsplatform --jdkhome /usr/lib/jvm/java-8-openjdk-amd64`

having both version 11 and 8 installed and having set openjdk11 as default

breiler, 2019-12-12:
I'm surprised that people are still having problems with this since we upgraded to Netbeans 11.0. I've noticed problems with OpenJDK 13 which could possible be fixed with Netbeans 11.2. I haven't had the time to experiment with it yet.

I'm using UGS on Debian 10.1 and OpenJDK 11.0.4 which works fine.

There is also an experimental version with a bundled OpenJDK 12 which you could try out here:
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT-linux.tar.gz

breiler, 2020-03-02:
Please try the versions with bundled Java from here: 
https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2020-03-02 by breiler

J-Dunn, 2021-10-31:
I'm not sure why this got closed. It does not help in finding this bug report. 
Certainly not fixed.  java 11 on up todate Fed34 crashes without showing anything on screen nor a word on console. 

If nothing else is dependant on java installing earlier version and switching system to use java 8 seems to fix initial problems running UGS classic. 

```
dnf install java-1.8.0-openjdk
alternatives --config java

```

breiler, 2021-11-01:
I closed the issue because we now bundle a Java version together with the Platform edition to mitigate the problems with users trying to run UGS on all sorts of JVM:s and OS. In theory that should work, but for different reasons it doesn't and we don't want to spend the little time we have on debugging this. 

If you are running the latest UGS Platform and are still experiencing a problems with it not starting, please post a new issue I'm happy to hear about it.

Regarding the Classic edition and older versions of UGS Platform requires specific versions of JVM on different OS and it would simply take too much time from us to create install instructions for all of them. On the install page we refer to Java 8: https://github.com/winder/Universal-G-Code-Sender/wiki/Installing which seems to work in your case as well...

J-Dunn, 2021-11-01:
Thanks, I appreciate the need to focus resources. Since there is a broad issue here and at least some work arounds exist, this should be put in a visible centralised place, like the install page you linked to.  I had already seen those instructions but 
"Make sure your system is using the correct JDK:" is not helpful.  You mention your bundled version here on a closed bug but not on the install page.

The very useful info provided above by carneeki could be put there too. I scanned three pages of recent bugs to look for similar issues but did not find this because it is closed. 

There are too many issues with "platform" for it to be a useful solution for  me. I have posted details of these bugs/shortcomings. 

The show stopper for me is disabling the command input until the machine returns to idle. I do fair amount if direct control for simple jobs like a small face milling operation where it is not worth going over to the CAD machine to make a 3D model, create a path job , create a gcode file for something I can type in by hand. 
Not being able to type the next command while the machine is cutting the last one, increases job time by a factor of two or three and reduces reliability. 

I also need to focus resources.