winder/Universal-G-Code-Sender #1152: ## Feature request- clearpath servo compatability

Guuswezel, 2018-11-24:
## Feature request
Would be great if this software would be Teknic Clearpath servo compatible, since these are upcoming motors in the hobby industry. The sFoundation files for these motors are downloadable from the teknic.com. Would love to hear whether this is possible or not.

winder, 2018-11-25:
This would be a feature request for GRBL/TinyG. UGS doesn't interface directly with the motors.

closed 2018-11-25 by winder

carneeki, 2018-12-16:
Drivers are a little different to controllers which ClearPath still requires. The flow goes a little like this:

CAM Software --> gcode
gcode -> UGS
UGS -> controller
controller -> driver
driver -> motors

The controller is what typically maintains the cartesian coords of the cutter position, and tells the drivers to rotate / reverse / stop.
The driver is typically an open loop system that just moves a stepper motor based on enable/pulse/direction signals. ClearPath differs slightly in that the driver has a little bit of smarts to move a DC motor with a built in encoder to a desired angle, which is how high speeds can be delivered, as well as high holding torque for a stationary shaft, and how it will attempt to correct the shaft position without any alignment. The ClearPath drivers (to my knowledge) don't take the Cartesian system into account. UGS doesn't (really) do this either, rather, it just asks the controller (eg GRBL/TinyG) where it thinks the cutter is in XYZ space.

Can this be implemented in UGS? Probably. But it would be a huge task because it involves writing a controller as a part of UGS to track the XYZ position and then tell the drivers what angle to set the shafts to, and what speed. It deviates quite a bit from the main focus of UGS: getting gcode from a file on a computer and into a controller (and also to act as a pendant and user interface to the controller).

I'd highly recommend trying out the Arduino/MSP430/Pi/BeagleBone/ARM type microcontroller and microprocessor forums, I wouldn't be surprised if someone is already working on a controller to do just what you're after, and if so, streaming to that controller is where UGS can shine. There might even be a GRBL port that rips the guts out of the enable/step/direction stuff, and replaces it with ClearPath controls. If so, I'd bet that UGS can already talk to that! :)

Guuswezel, 2018-12-16:
Thanks for the reply and explanation!

carneeki, 2018-12-16:
For anyone interested in writing a controller, I've spent < 10 minutes and found a few examples in the sFoundation SDK for running single and multithreaded motion control with the use of their SC4 hub.

The controller could sit on the same PC (or a separate one like a Pi) as a piece of software, and take G-code via TCP port from UGS, and use this to convert G-code into the motions required. It won't be trivial to implement, but at first glance it looks like a pretty good starting point.

Guuswezel, 2018-12-16:
I could also use the A and B ports of the motors to connect a motion control board to them, so I could overcome this problem. But by doing this, all the benefits of these servos are worthless. So if anyone is willing to write the controller, please hit me up so we could make a deal since I save loads of time and money trying the other method