winder/Universal-G-Code-Sender #1153: Universal-G-code-sender resets grbl when issued Z command

atyanands, 2018-11-25:
When i send  z axis command my GRBl resets.
This does not happen with anything else.
This is the output code: 

**** Connected to COM3 @ 38400 baud ****

Grbl 0.9j ['$' for help]
>>> G20 G91 G0  Z10
ok
<Run,MPos:
Grbl 0.9j ['$' for help]
>>> G91 G0  Z10
ok

Grbl 0.9j ['$' for help]

Can anybody tell me what might be wrong

AbySet, 2018-11-26:
Hey;
- Please, help us to help you, there is a template for issues whitch provide us with more infos about what OS you are using, what hardware, what version of program and so on, would be helpfull if you post according to the template, these informations are needed so we can figure out what's wrong and give back some help.
- According to your post, this is a grbl issue not ugsplatform but that's ok.
- 38400 is not a baudrate known to be used with grbl 0.9 but 9600 is (if i'm not mistaking).
Thank you.

AbySet, 2018-11-26:
Good to know and thank you so much for reporting how you did fix it, this may help other people.
I have learned that big steps/mm can reset grbl and that 0.9 version of it can run on 38400.
Enjoy your CNC.

winder, 2018-11-27:
Thanks @AbySet