winder/Universal-G-Code-Sender #1154: Z Plane (YZ) Arc problem

dustnnotes, 2018-11-27:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

dastultz, 2018-11-27:
There's pretty little to go on here, but this might be a duplicate of https://github.com/winder/Universal-G-Code-Sender/issues/853

dustnnotes, 2018-11-27:
Thanks for the quick response @dastultz!
Sorry, I had an issue uploading the G-code the first time, posted now.  I took a look at #853, but wasn't sure, and wanted to provide my case as well.  This is a blocking issue for me on this for operation, and I would rather not use multiple senders.

AbySet, 2018-11-28:
I tested it even with 11.25.2018 version nightly build and confirm the issue.

Tested under:
-Windows 10 x64
- Java 8.191.

dastultz, 2018-11-28:
Yes, sorry, I meant to mention disabling Arc Expander. Couldn't remember if I enabled it or disabled it as a work around.

DanielWeigl, 2019-04-26:
This looked like a nice bug to get to know the codebase a bit better.. Looked into it and I think I found it, fixed in https://github.com/winder/Universal-G-Code-Sender/pull/1220

Tested against robomechs and dustnnotes GCode:    
[image]

[image]

dastultz, 2019-04-28:
Awesome, I'll try to confirm after merge. Thanks.

commit 82ec70cf (2019-04-29): Merge pull request #1220 from DanielWeigl/#1154_errors_in_ArcExpansion

fixes #1154 - fix the radius-calculation to respect the current selected working plane
commit b274526f (2019-04-29): fixes #1154 - fix the radius-calculation to respect the current selected working plane in  `addArcPointSegment`

also initialize the starting-Point with the correct Unit (it was not really triggering the bug, but it was throwing me off in the wrong direction, and might cause other bugs in the future)
closed 2019-04-29 by winder