winder/Universal-G-Code-Sender #1155: Unknown units in autoleveler receiving probe.

QuantumKittens, 2018-11-27:
While using the autoleveller I randomly get a popup (after probe contact) with "Error while parsing response" and "Unknown units in autoleveler receiving probe." in the console. The probe cycle continues normally, but no probe data is saved.

It seems that the probe response from grbl gets messed up somewhere along the way and that the newline is lost:
 [PRB:1.500,3.000,0.244:1]2ok

I have not found a reliable way to reproduce this behavoir, it occurs randomly. (Seems to be more likely on an arduino clone though.) It does only appear with UGS, not when communicating directly with grbl using a serial terminal.

A (quick / temporary) fix might be to change the regex that parses the probe response so that the brackets are used as a delimiter. (Currently this seems to be the newline.) This would ignore any following erroneous characters. 
#### Version

  UGS Platform 
   stable and Nightly Build as of November 27, 2018 (the later not tested extensively, but seems to be affected too)

#### Operating system
MacOS
Windows 7 (only tested with stable)

#### Platform
  GRBL 1.1f
  MoonCactus STM32 grbl (More errors than with an Arduino.)

AbySet, 2018-11-28:
Hey,
The autoleveler currently does not work properly, you can see the message in the App, may be under dev right now, i'm just a user like you and i would wait for some improuvement about it.

QuantumKittens, 2018-11-28:
For me it was actually working fine, but only once. Might have been luck.
Anyone wanna have a quick look at https://github.com/winder/Universal-G-Code-Sender/blob/b35c908ebda63953e8d612018d16b3c923119423/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java ? (The interesting stuff seems to be around line 273)
It may work to just ignore any characters after the closing bracket ']' of a grbl report.

phcreery, 2018-12-30:
It worked one on mine also

winder, 2018-12-31:
I merged in #1166 from @JulioMCDias which improves unit detection with the autoleveler. Unfortunately I still haven't had time to look into the broken nightly build process, so to get the change you'll need to use the source: [details here.](http://winder.github.io/ugs_website/dev/getting_started/)

winder, 2018-12-31:
@QuantumKittens you might be right about the newline though, in that case there would still be a problem. Are you using an Arduino clone? They are known to have issues with the serial port.

QuantumKittens, 2019-01-27:
I did test it with a genuine Arduino, a clone and also with an STM32 port of grbl. (https://github.com/MoonCactus/grbl-STM32F103) The issues were comparable with all three. I don't have a working setup at the moment (switched to LinuxCNC for now), I will check it in around two weeks.

breiler, 2023-08-18:
Please try with version 2.0.19, make sure to double check your settings after upgrading:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.0.19

I am closing this as the issue is very old, if you find any problems please create a new issue.

closed 2023-08-18 by breiler