winder/Universal-G-Code-Sender #1156: Keymap Import stuck

mechanicalgoose, 2018-11-29:
## Problem description 
With the latest Nov 25 nightly build, when attempting to import a *.zip folder for a new keymap set, the dialog screen gets hung on "Loading..." for extending periods. The *.zip folder uploaded without issue in previous releases. The previous version I had before today was Nightly May 2018 and it does not exhibit this issue.

I confirmed I have the latest Java also. 

Thanks.

  UGS Platform 2.0 - Nightly Build Mar 23, 2018

#### Operating system
  Windows 10

winder, 2018-11-29:
Is there anything in the logs? You can also try clearing out the cache directory. Details here:
https://winder.github.io/ugs_website/guide/troubleshooting/#property-files

This is one of the many feature that come from the Netbeans Platform framework used by UGS, so I'm not very familiar with how it works.

winder, 2018-11-29:
The email-reply feature strips attachments, you'll need to attach the file on github.

mechanicalgoose, 2018-11-29:
Ah, ok. Will do that when I get home this evening. 

-JW

AbySet, 2018-11-29:
The issue is actualy in "Select option for import/Export", seems like the options list is stuck and does not load what it should show, i don't have a zip file to test with but it's doing the same for export function.

mechanicalgoose, 2018-11-29:
Exactly. It does not populate the list for user selection to import. It just displays ”Loading...”

-JW

mechanicalgoose, 2018-11-30:
Attached is my log file I attempted to share this morning.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/2631416/messages.log)

winder, 2018-11-30:
The issue seems to happen when exporting settings as well... I see this exception (locally, with your log, and with java8/11). It might mean one of the UGS settings is empty/missing:

```
java.lang.NullPointerException
	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
	at java.lang.Double.parseDouble(Unknown Source)
	at org.netbeans.modules.options.export.OptionsChooserPanel.createOptionsTreeModel(OptionsChooserPanel.java:438)
	at org.netbeans.modules.options.export.OptionsChooserPanel.access$700(OptionsChooserPanel.java:95)
```

Fixing it would require more time for troubleshooting than I have right now.  If somebody want's to give this a try I'd suggest the following:
1) try disabling as many UGS plugins as possible, resetting UGS, then exporting. If it works you can narrow down which plugin is the problem.
2) Set a breakpoint in `OptionsChooserPanel`, hopefully in the "catch" and see which option is throwing.

AbySet, 2018-11-30:
Can'nt rearly help with this because i dont have that far knowlege but since the issue is known we can just wait till somebody or you have time to fix it.
PS: I should learn java programing someday.

mechanicalgoose, 2018-11-30:
The problem persists after deactivating all possible plugins.

carneeki, 2018-12-03:
I can confirm the same problem @mechanicalgoose is exhibiting, and have disabled all possible plugins.

I cannot find the string "OptionsChooserPanel" in the UGS source - I'm guessing it's part of the NetBeans source, but my lack of experience with mvn+NB is making this a bit trickier than I had hoped. I'm slicing through the revision history right now to see if I can isolate it.

carneeki, 2018-12-03:
Update: commit d50a29cd91f5078cacf51c635d70e92334888014 seems to be where the problem exists. I can build and get to the export screen with a9b19703492d6ca9a020b9b3a16724b60a41760a, but not the one following it.

The diff message "Moved where the version is set" and creates a function to set a system property called "netbeans.buildnumber". Looking at the diff between the two commits - it doesn't make much sense as to why this causes the problem, hopefully @breiler may know. It sounds like the property may not exist when the settings are trying to export.

winder, 2018-12-03:
@carneeki Nice find, never would have expected that to be the root cause. There were a lot of bug reports using the Netbeans version instead of the UGS version, so we were using this to update the "Product Version" string in the about page based on the recommendation here:
http://wiki.netbeans.org/DevFaqVersionNumber

The problem... ugh... it looks like the option exporter assumes that `netbeans.buildnumber` is going to be a simple date string. 

org.netbeans.modules.options.export.OptionsChooserPanel.createOptionsTreeModel(OptionsChooserPanel.java:438)
```java
        String nbBuildNumber = System.getProperty("netbeans.buildnumber"); // NOI18N
        try {
>>>         currentBuildNumber = Double.parseDouble(getOptionsExportModel().parseBuildNumber(nbBuildNumber));
        } catch (NumberFormatException nfe) {
            LOGGER.log(Level.INFO, "Could not parse netbeans.buildnumber: {0}", nbBuildNumber);  //NOI18N
            currentBuildNumber = 201403101706.0;  // default to build date of 8.0 version
        }
 ```
Then `parseBuildNumber` fails here:
```java
    String parseBuildNumber(String strLine) {
        Matcher matcher = DATE_SIMPLE_PATTERN.matcher(strLine);
        if (matcher.find()) {
            String year = matcher.group("YEAR");
            String month = matcher.group("MONTH");
            String day = matcher.group("DAY");
            String hours = matcher.group("HOURS");
            String minutes = matcher.group("MINUTES");
            String time = (hours != null && minutes != null) ? hours.concat(minutes) : "2359";  // NOI18N
            return year.concat(month).concat(day).concat(time);
        }
>>>     return null;
    }
```

I tried a couple things, but so far haven't been able to get everything working.

carneeki, 2018-12-03:
Does the version number have to follow the `YYYYMMDDHHmm` format?

I'm wondering if perhaps the build script could populate the appropriate `.properties` and `pom.xml` files with the shortened commit hash for the source version in the documentation you linked to? I think the commit hash is probably more useful for different feature branches than the date, even if it seems a little foreign to users.

An alternative idea:

```java
if(matcher.find()) {
    // build the string
    return theString;
}
throw MatcherNotFoundException();
```

and

```java
catch(NumberFormatException | MatcherNotFoundException e) {
  // no change
}
catch(Exception e) {
  // log some other unhandled exception differently and set the build number stuff as before
}
```

breiler, 2019-02-21:
Sorry guys, I completely dropped the ball on this one.

I have a suggestion for a fix. It's not perfect but we can use a ISO-like (yyyyMMdd) date format instead of the version string used today. The about dialog wouldn't show the version number, only the build date:

[image]
[image]

closed 2019-02-24 by breiler