winder/Universal-G-Code-Sender #1163: Do we have already something for surface scanning?

krtschmr, 2018-12-08:
## Feature request
I need to be able to go from a point 5cm into each direction and probe down the Z axis and get a list of all coordinates with 1mm spacing on X/Y axis.

easy surface scanning. does this already exist as a feature?

if not, should it be a new module/plugin or hijacked into the current probing module?

AbySet, 2018-12-10:
You may put this into Pull requests section to get more chance for aswer. (Next tab on this screen).

carneeki, 2018-12-10:
Pull requests are typically for code to be pulled in and merged from forks
of UGS. I recommend asking an admin to tag it as a feature request.

It might sit as a todo list item until someone can write some code and then
send a PR to close this issue. I know I have a few things outstanding to
look into before I can look into it.

I think a surface keeping module would be totally sweet though! I imagine
it's going to be something like doing a 2d grid of G38.2 ops and storing
the Z position. CSV keeps it portable.

winder, 2018-12-10:
You might be able to do this with the existing "AutoLeveler" module. It lets you control the grid size and resolution pretty much exactly the way you described and there is a "View Data" button which provides the coordinates as JSON array. Be careful, this feature was never fully tested.

Implementation wise, there is a probe cycle which collects the data, then a gcode processor hook to apply the heightmap to gcode. So if this is a common task people need the auto-level plugin could be split in 2 so that it's easier to generate/access the heightmap.

https://www.youtube.com/watch?v=n8P9pl-2Lmk
https://www.youtube.com/watch?v=2gEy3Wjo1tU

If you use this and it works, please let me know :)

closed 2018-12-10 by winder

AbySet, 2018-12-10:
Yes **carneeki** you are right, i did a mistake there, so sorry **krtschmr**.
I'll pay more attention next time.

krtschmr, 2018-12-11:
@winder  i didn't find that feature. i looked into the code and somehow i don't understand all of it (i'm just a ruby dev). i'm not sure if i want to *try out* that feature. i bought a touch probe and destroyed it yesterday, i still have no clue why. it didn't came up high enough and broke off. now i have to wait 2-3 weeks for a new one. trying out might be a bit *dangerous*. don't wanna risk again ;)

winder, 2018-12-11:
Fair enough.

Don't sell yourself short though. The modular framework UGS uses is pretty complex, but if you're interested and focus on an individual module I bet you can work through it. If you do take another look, the main probe routine is [HERE](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-surfacescanner/src/main/java/com/willwinder/ugs/platform/surfacescanner/AutoLevelerTopComponent.java#L540). It loops over the x/y/z coordinates of each probe locations, the same data structure is used to draw the dots in the visualizer.

The probe result is returned asynchronously in the [UGSEvent callback](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-surfacescanner/src/main/java/com/willwinder/ugs/platform/surfacescanner/AutoLevelerTopComponent.java#L161) and is saved to the heightmap.

Almost everything besides those functions is GUI related.

rostrovsky, 2020-09-30:
Hi @winder, this is awesome feature.

Is there a way to apply saved heightmap JSON?
I have scanned my bed surface and would like to apply its heightmap from file to avoid scanning it before every job.