winder/Universal-G-Code-Sender #1165: Feed/Spindle speed override shortcuts do not function when code is being sent

jahnj0584, 2018-12-15:
### Specifications
#### Version
  UGS Platform 2.0 - Nightly Build 

#### Operating system
  Windows 7

#### Platform
  GRBL 1.1f

carneeki, 2018-12-20:
The expectation is that overrides should be sent realtime, that is, they should "jump to the head of the queue" with UGS' internal buffer and be streamed to the controller (eg GRBL) immediately. (if UGS is in the middle of sending a command, it should complete that command and then send the feed override, otherwise strange things may happen).

According to the GRBL manual ([do a find for "Feed overrides:"](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands)), GRBL prioritises realtime commands (like feed+speed override) and processes them instantly. This makes sense to do it this way for things like feed hold, but also for feed + speed overrides, here's a use case as to why:

Pretend we start at X0 Y0 and run `G1 X400 F100`
and we encounter some chatter, and we want to increase feed rate to overcome the harmonic. Sending a feed override increase of 1% (ie UGS would send a `0x93` byte to increase the feed rate by 1%) and GRBL will start to run the stepper 1% faster (now 101mm / minute).

For GRBL, this should be true for *all* real-time commands, including the necessary ones @AbySet mentions. I think the same would be true of TinyG and Smoothie as well. I was looking at some of the button code last week, and I saw that UGS has some buttons which should send a command immediately using `IController.sendOverrideCommand(Overrides)` (by way of `GUIBackend.sendOverrideCommand(Overrides)`).

@jahnj0584 - do you see anything in the command line to show if UGS attempted to send an override? If not, it could be in here that the problem is occurring (the `if(realTime...` line here near line 774 of `GrblController.java`):
```java
    public void sendOverrideCommand(Overrides command) throws Exception {
        Byte realTimeCommand = GrblUtils.getOverrideForEnum(command, capabilities);
        if (realTimeCommand != null) {
            this.dispatchConsoleMessage(MessageType.INFO, String.format(">>> 0x%02x\n", realTimeCommand));
            this.comm.sendByteImmediately(realTimeCommand);
        }
    }
```

jahnj0584, 2018-12-21:
I wonder if this is actually just a UGS issue. The built in button for over rides work. If you use a keyboard shortcut, it won't.

dgjohnson, 2019-03-26:
I confirmed the behavior that the spindle and feed override hotkeys do NOT work when gcode is being sent.
is there any workaround for this?

breiler, 2019-03-26:
I think that I found the cause so I added a bug label. The service handling shortcuts checks if the machine is idle. Otherwise the command is not sent.
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugscore/src/main/java/com/willwinder/ugs/nbp/core/control/RunActionService.java#L50

dgjohnson, 2019-03-26:
that may be to systemic to be correct, as other hotkeys work when the machine is not idle...

dgjohnson, 2019-03-26:
will buy beer for new build with fix! :)

dgjohnson, 2019-03-28:
> I think that I found the cause so I added a bug label. The service handling shortcuts checks if the machine is idle. Otherwise the command is not sent. [...]

by the way, where is this called when processing the hotkeys?

breiler, 2019-03-28:
The actions get registered here:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugscore/src/main/java/com/willwinder/ugs/nbp/core/control/RunActionService.java#L82

dgjohnson, 2019-03-28:
thanks, i'm new to NetBeans, I'm going to compile locally and test the change.

closed 2019-04-10 by winder

breiler, 2019-04-11:
> good catch, will buy beer for a pull request! :)

Looking forward to my beer @dgjohnson :)

dgjohnson, 2019-04-11:
> > good catch, will buy beer for a pull request! :)
> 
> Looking forward to my beer @dgjohnson :)

no problem! tell me where to send it... david@diy.engineering

breiler, 2019-04-12:
Ha ha, I was only kidding. Just pay it forward!

dgjohnson, 2019-04-12:
[image]

PR #1210 (2019-04-10): Fixed bug on shortcut commands for overrides 
Fixed bug where shortcut commands for overrides couldn't be executed when machine was running.  Also checks the controller for real time override support.

Fixes #1165