winder/Universal-G-Code-Sender #1167: Ugs Platform not loading

Woodmonger, 2018-12-20:
Ok still pretty new to the ugs but do have a pretty good hang of things. 

UGS opened normally about 4 days ago and ran perfectly. I now open it and the splash screen shows up then disappears and nothing happens after that. I have downloaded the latest build and still the same thing happens.

Any help would be appreciated.

AbySet, 2018-12-23:
Can you please use the template for issues so we can know what exactly you have installed, we don't even know on what OS you are working ! The template will provide us with needed infos so we will be able to help you, like OS, Java version, and  so on.
Thank you.

carneeki, 2018-12-23:
Also checkout the troubleshooting page on GitHub and Will Winder's site,
the answer may be there too.

breiler, 2019-02-24:
Closing as there isn't enough information to investigate.

closed 2019-02-24 by breiler