winder/Universal-G-Code-Sender #1169: Visualizer works w/Platform but not Classic

Signdude, 2019-01-05:
Thanks for any help.

## Problem description 
Visualizer works fine with Platform Version, just gives a blank white window in Classic version.

### Expected Behavior
After loading a gcode file using Classic... I expect Visualizer to deliver a representation of the gcode.

### Actual Behavior
After loading a gcode file using Classic... Visualizer gives a plain, white, empty window.

### Steps to Reproduce the Problem

  1. Open the program
  2. Open a gcode file
  3. Select Visualize
  4. The Visualizer presents a Blank Visualizer window.

### Specifications
#### Version

Universal Gcode Sender Classic (Version 2.0(nightly)/Dec. 18, 2018)

#### Operating system
Windows 10 (64 Bit)

#### Platform
  GRBL 1.1f
  Java 8 Update 171 (64 Bit)
  

#### Other
[image]

AbySet, 2019-01-06:
Try to update  java, i think 171 version is outdated.

Signdude, 2019-01-08:
Thanks for the note. Just uninstalled 8u171 and installed 8u191. Visualizer still blank.

Thanks again.

AbySet, 2019-01-08:
I believe it's using OpenGL or D3D, updating video card driver and directX may help. Not quiet sure but i would do it.
This may have nothing to do with, but i also would clear the cash files, they are located at:
c:\users\<UserName>\.ugs

Signdude, 2019-01-09:
Thanks again.  I **do** appreciate your attempts to guide me in the correct direction, but I have no idea what OpenGL or D3D are.  

I updated UGS to Version 2 primarily to take advantage of the new GRBL 1.1's ability to use my probe for my Shapeoko XXL CNC.

The result of upgrading both is... not only does my probe **still** not work, my keypad pendant doesn't work either and now I have no visualizer.

So now I have what was once a working CNC sitting idle.  I guess I'm screwed.  I'll have to find a way to reflash my controller to GRBL 0.9 and find and download the orginal UGS.

Thanks again.

AbySet, 2019-01-09:
I will download and use classic version to test for you, i keep you informed. But you can go to previous version if you have work waiting to be done.

AbySet, 2019-01-11:
Hey,
I have tested classic version and i have same issue (blank white visualizer) so i think it's an issue that need to be fixed.
My config:
Windows 10 x64.
grbl 1.1
java and video drivers all up to date.
 I hope the owner will fix this.

But i still have a question:
Is there any reason for using classic version? platform version seems better to me.

JacksCrafts, 2019-01-12:
Hey Signdude,
I agree with AbySet, you should try the platform.
You can move the mudules around to where you like.
[image]
The macros are easy to set up. The probe module does not work for me. And the jog buttons in the lower right corner are not prcise (you have to get your mouse in just the right spot) and if you change from MM to Inch and back to MM the number is not the same. So there are some kinks that still need worked out. But I like it very much. Once I got the camera mod I will never go back. 
[image]
I use this everday, Thanks Winder

Signdude, 2019-01-12:
Thanks for the note Jack.  I downloaded and installed both Classic & Platform after using the original UGS for a couple years.  Since I was already familiar with the original I tried the Classic version first.  The first problem I ran into was the problem with the Visualizer.

So... I loaded up the Platform version and gave it a shot.  I loaded a small file then checked the Visualizer and thought, well this looks like the way to go.  I setup my piece, homed the gantry, zeroed everything, removed the bit for an air cut and hit send.  

The gantry began heading at rapid speed toward the home position, paused and began all kinds of movements that had nothing to do with the job I was sending.  I immediately killed power before I had any physical problems occurred.

I homed, repositioned, re-zeroed, re-visualized and hit send.  It cut the job correctly.  I have no idea where the mistery gcode came from.  No other jobs had ever been run with the Platform version prior.  

I've done some reading on the Platform version and it **does** look like it has some great new features.  I've had a fit trying to get the various windows positioned to my liking, but I'll eventually get a combination that I can live with.

Thanks again for the feedback.  I'm going to give it another going over this weekend.

Bill

Signdude, 2019-01-17:
Thanks AbySet.

With regards "right-mouse jogging"...  Until you mentioned it I didn't realize that was a feature.  However, using the right mouse button has no effect on any direction on my machine.  Wondering if I'm the only one having the problem.

Thanks again.

p.s.  I'm still having a couple problems getting my windows arranged the way I'd like them.  Primarily the issue I have is with both the "Probe" and "Macro" windows.  They seem to have a fixed width which makes it impossible to fit either of them under/above a narrower window.  I'm going to begin searching Issues to see if there is a reference to this by someone else.

Signdude, 2019-01-17:
Accidental Close.

AbySet, 2019-01-17:
No man you'r not the only one that right mouse don't work, as soon as i saw your post i ran to test it and it's not working for me anymore, i have no explaination about that.
(after a min)
I still have older versions of UGS (oct 21 2018) and i tested the feature with it and was working seems like they removed it.
One mistake i did in the last post, you have to left click on the oposit direction to stop it not on the same direction.
Macro window is fully movable and resizable to me.
Probe window however is movable and resizable but i confirm that it has a small issue with the width, you can not make it smaller than a certain value (whitch still prety large for the side pannel).
You need to detache them from theire bottom location before you get able to resize them, to do so, you drag them from their titles.

Signdude, 2019-01-17:
Hey AbySet,

Thanks for the note.  You're correct about both Probe and Macro windows.  I was eventually able to get the Macro window scaled down to where I can use it.  The Probe window however, can only be contracted vertically.  I can move it anywhere, but if I move it to a narrower area all windows in that area expand to match the size of the Probe window and can't be changed after that.

Thanks again for the heads-up on the jogging issue.  If they find that the issue is not just my system and make a fix to it, I'll be prepared to left-click the opposite direction to stop the jog.  :)

Thanks again.

Signdude, 2019-01-19:
Hey Guys , 

Thanks AbySet, JacksCrafts and Will,

At your and others recommendations, I have accepted the inevitable and am using the Platform version of UGS.

I really wish someone could figure a way to get the Visualizer in the Classic version to work.  I’m sure there are others who would prefer Classic.  I’m not a big heavy maker, but I do enjoy using my Shapeoko XXL frequently.  I don’t really need all the new features of the Platform version.  I’m probably not going to need the Override functions, I have no idea what Proxy Settings are and I’ll probably never use the Dowel Maker or the UGS Surface Scanner functions.  There’s no doubt that Platform is a powerful and well thought out program, but at my stage of progression in CNC it really has my head spinning a bit just trying to do some fairly simple jobs.  Most of my jobs are nails... a Sledge would be overkill where I just need a hammer.  

I’m sure you’re probably familiar with GRBL Panel.  I just finished giving it a look over.   I love it’s simplicity.  It’s a plain vanilla program similar to the original version of UGS.  Unfortunately, it doesn’t have a Visualizer.  I really need one to make sure the file I opened is the one I intended to send to my machine.  

Don’t get me wrong... I am getting more comfortable with arranging/organizing all the windows and I am able to make chips with my router.  But, for me...   I would be very happy with Classic, but I have to have a Visualizer. 

Thanks so much for your help.

Bill

From: AbySet 
Sent: Friday, January 11, 2019 5:10 PM
To: winder/Universal-G-Code-Sender 
Cc: Signdude ; State change 
Subject: Re: [winder/Universal-G-Code-Sender] Visualizer works w/Platform but not Classic (#1169)

Hey,
I have tested classic version and i have same issue (blank white visualizer) so i think it's an issue that need to be fixed.
My config:
Windows 10 x64.
grbl 1.1
java and video drivers all up to date.
I hope the owner will fix this.

But i still have a question:
Is there any reason for using classic version? platform version seems better to me.

JacksCrafts, 2019-01-19:
Hi Bill,
You might try Source Rabbit, it is free open source. 
JacksCrafts

AbySet, 2019-01-19:
I can see you have prety good reason for using classic version, everybody has his own business, i still confident that they will fix it.
I'll keep trying some stuff around for you and let you know if i get something positive.
Wish you the best.

Signdude, 2019-01-19:
Thanks Guys,

I just downloaded and installed Source Rabbit on my "house" computer.  I have to be connected to my CNC before it will open.  So, I'll be carrying my flash drive out to the shop this afternoon and give it a spin.

Can't thank you guys enough for sticking with me.

All the best to you and yours.
Bill

Signdude, 2019-01-20:
> 
> 
> Hi Bill,
> You might try Source Rabbit, it is free open source.
> JacksCrafts

Hey  Jack,

Well....  I gave it a spin.  :)  What a cute little program.  :)  That's about as "plain vanilla" as you can get.  But, it does perform the basic functions to load and cut a file.  Thanks.

I'm staying with Platform... until someone can find a way to get Classic's Visualizer working for me.

Your help is much appreciated.

BIll

breiler, 2020-12-09:
This will be fixed with the jogamp 2.4.0 version (still waiting for it to be released)
I downloaded it manually and tried it out and it looks like it is working on MacOSX at least.

breiler, 2023-02-01:
We have upgraded to Jogamp 2.4.0 in version 2.0.13 so I'm closing this.

closed 2023-02-01 by breiler

papillion47, 2024-10-29:
3-4 senedir tıkır tıkır çalışmakta olan GRBL programıyla çeşitli işler yapabilmekteyken, 1 aydan fazla zamandır tüm uğraşmalarıma rağmen GRBL visualizer halâ çalışmıyor. Ben CNC ve g kodu çalışmalarında çok usta biri değilim. Bu arada defalarca GRBL yi kaldırıp yeniden yükledim. CNC kartını sipariş edip yeniden yükledim. Konsol dan yararlanarak olması gereken kodları yeniden yükledim. Jog komutları, spindle, diğer fonksiyonlar doğru çalışıyor. GRBL Hiçbir hata mesajı vermiyor.. Sadece "visualizer" görev yapmıyor. Rastgele bir tap uzantılı basit bir dosya açıyorum. Visualizer de görünmüyor ama CNC de işleme başlatılıyor. Normal çalışıyor. Bunalım içindeyim.. Bu derdime bir çare gösterebilecek olan yok mu?..