winder/Universal-G-Code-Sender #1170: Probe Helper: Corner probing not showing up

splobucket, 2019-01-06:
Probe helper doesn't show the buttons to do an outside probe anymore? The probe helper window shows the feedrate, plate thickness, probe distance/direction, and axis. Axis only has x, y, and z option.

I dont see anywhere where i can do an XYZ probe like I used to be able to do.
  
UGS Platform 2.0 - Nightly Build Mar 23, 2018
Linux Bunsenlabs
  GRBL 1.1f
[image]

winder, 2019-01-09:
You have an old build. The original probe helper only had that simple Z probe option.

closed 2019-01-09 by winder

splobucket, 2019-01-09:
> single-axis probe option.

How do I get the new build? I downloaded the stable build (2.0) from this
page: https://winder.github.io/ugs_website/download/.